VUnit/vunit #710: check_equal hex output

Munken, 2021-03-05:
I'm currently writing test benches for code that handles 32 bit addresses. 
In my test bench I'm writing the addresses a hex string eg. X"deadbeef".

However, when a check_equal fails it will output the current and expected result in base 2 and 10.
```
Got 1110_0000_1101_1101_1111_0000_0000_1101 (3772641293). Expected 0001_0100_1010_1011_1111_1101_1111_1011 (346815995)
```
Would it be possible to either also print the result in base 16 or have the number printed in the parenthesis in a user chosen base?

 ```
Got 1110_0000_1101_1101_1111_0000_0000_1101 (3772641293/0xE0DDF00D). Expected 0001_0100_1010_1011_1111_1101_1111_1011 (346815995/0x14ABFDFB)
```

LarsAsplund, 2021-03-06:
This has been discussed before but I don't think we have a previous issue for it. The current format is a bit of a compromise between being able to have the ideal format for the data being checked and not having too long messages. The "nibble format" serves two purposes. One is to see the individual bits if that's what matters for your application (e.g. control bits in a register). The other purpose is see the hex format but it assumes that you can do the nibble to hex number in your head.

To avoid the long messages I propose that we don't add another format. We could have a configuration for the checker but the problem with that is that the format would be the same for all calls. A third approach would be to control the format in each call.  Today we have the `result` function allowing you to control the auto generated message. Maybe we can add another parameter to that function. For example

`check_equal(my_data, expected_data, result("for my_data", hex))`

 Any thoughts?

Munken, 2021-03-07:
I would agree that simply expanding the already long log message could lead
to trouble.
And what should be done when someone requests the next format?

I thus agree with you that expanding the arguments for the result function
is the best option.
This allows for maximum flexibility and can be user expanded.

This has been discussed before but I don't think we have a previous issue

nselvara, 2025-07-03:
I'd second this feature.

Having a longer message which only pops up during failed `check_equal` checks is perfectly fine in my opinion, especially since testbenches that use `check_equal` usually won’t have intentionally failing checks anyway.

I'd also be fine if there were an option to let the user specify their preferred numeric base or formatting. That way, people can choose whether they want hex, **_binary_**, **_decimal_**, or a **_combination_** in the failure message.

nselvara, 2025-08-25:
Maybe we could solve this cleanly by introducing a `show_style` setting (enum/type) similar to `show(display_handler, ...)`. That way the user can pick between styles like `hex`, `dec`, `bin`, or combinations without changing the core check logic.