VUnit/vunit #711: ModelSim License not supported

rafaelnp, 2021-03-17:
I am getting an error, when running VUnit in tadem with ModelSim, the license is not recognized by ModelSim.
I have additionally searched for information in Intel documention and the license is configured correctly and It can be executed without VUnit. The enviroment infos follow below.

I suspect, some additional parameter, that should be passed to ModelSim,  is missing. Did anybody has already seen this error? Any thoughts?  Thanks in advance.

Version Infos:
| Software    | Version |
|------------ | ------------ |
| ModelSim    |  Intel FPGA Edition 2020.1 |
| Operating System | Windows 10.0.19041.867 |
| Python         |  3.9.2 |
| VUnit           | 4.4.0 |

VUnit output:
```bash
> python.exe .\run.py
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd                        passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\path\src\path.vhd                                    passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\print_pkg.vhd                            passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\types.vhd                             passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec_builder.vhd                     passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec_builder-2008p.vhd               passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec.vhd                             passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec-2008p.vhd                       passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_string_pkg.vhd           passed
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_string_pkg.vhd(40): Function 'read_char' may complete without a RETURN at line 39.
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_string_pkg.vhd(46): Function 'get_ptr' may complete without a RETURN at line 45.
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\string_ptr_pkg.vhd                    passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\ansi_pkg.vhd                             passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_levels_pkg.vhd                       passed
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_levels_pkg.vhd(33): (vcom-1246) Range 1 to 0 is null.
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\string_ptr_pkg-body-2002p.vhd         passed
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\string_ptr_pkg-body-2002p.vhd(289): Function 'to_string' may complete without a RETURN at line 287.
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\byte_vector_ptr_pkg.vhd               passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_integer_vector_pkg.vhd   passed
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_integer_vector_pkg.vhd(40): Function 'read_integer' may complete without a RETURN at line 39.
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_integer_vector_pkg.vhd(46): Function 'get_ptr' may complete without a RETURN at line 45.
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\integer_vector_ptr_pkg.vhd            passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_handler_pkg.vhd                      passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\print_pkg-body.vhd                       passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\logger_pkg.vhd                           passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\dictionary\src\dictionary.vhd                        passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\run\src\run_types.vhd                                passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\file_pkg.vhd                             passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_handler_pkg-body.vhd                 passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\integer_vector_ptr_pkg-body-2002p.vhd passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\integer_array_pkg.vhd                 passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\queue_pkg.vhd                         passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\string_ptr_pool_pkg.vhd               passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\queue_pkg-body.vhd                    passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\queue_pkg-2008p.vhd                   passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\integer_vector_ptr_pool_pkg.vhd       passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\run\src\runner_pkg.vhd                               passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\run\src\run_api.vhd                                  passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\queue_pool_pkg.vhd                    passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\integer_array_pkg-body.vhd            passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\dict_pkg.vhd                          passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\data_types_context.vhd                passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\core\src\stop_pkg.vhd                                passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\core\src\stop_body_2008p.vhd                         passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\core\src\core_pkg.vhd                                passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\run\src\run.vhd                                      passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\logger_pkg-body.vhd                      passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_levels_pkg-body.vhd                  passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_deprecated_pkg.vhd                   passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\check\src\checker_pkg.vhd                            passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\run\src\run_deprecated_pkg.vhd                       passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\vunit_run_context.vhd                                passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\check\src\checker_pkg-body.vhd                       passed
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\check\src\check_api.vhd                              passed
[... 47 lines cut ...]
```

and the testbench file test_tb.vhd:

```vhdl
library vunit_lib;
context vunit_lib.vunit_context;

entity test_tb is
	generic (runner_cfg : string);
end entity test_tb;

architecture rtl of test_tb is
begin
	main: process
	begin
		test_runner_setup(runner, runner_cfg);

		wait for 10 ns;
		check_equal(14, 14);

		test_runner_cleanup(runner);
	end process main;
end architecture rtl;
```

eine, 2021-03-17:
This is the relevant part of the log:

```
# Start time: 12:47:15 on Mar 17,2021
# ** License Issue: License server does not support version 2020.02 of the requested feature(s).
# ** License Issue: License file does not support this version. (C:\intelFPGA_pro\20.1\license.dat)
# ** Error: Failure to obtain a VHDL simulation license. Unable to checkout 'alteramtivsim' license feature.
# Error loading design
```

Why is it using 2020.02 and checking the license in `C:\intelFPGA_pro\20.1\license.dat`? Do you have multiple instalations of ModelSim/QuestaSim? Do you have any envvars set up for pointing to the license (server)?

rafaelnp, 2021-03-17:
@eine: I have Modelsim 2020.1 with a perpetual license and Intel Quartus Prime Pro 2020.1. Both  node locked licenses. Additionally there is no license server. I turned off the NIC and disconnected the UTP cable and the resul is the same.

[image]

eine, 2021-03-17:
Can you try pointing directly to the path of the installation you want to use (http://vunit.github.io/cli.html#simulator-selection) and pointing to the license file using the envvar supported by it (maybe `LM_LICENSE_FILE`)?

rafaelnp, 2021-03-23:
@eine I dug into the issue and discovered the previous version (2019.2) installed, was not uninstalled properly and it was being called by vunit using the new license. After properly removing theses files and restart the computer, it worked as expected (see below). 
Just for completness, the Modelsim searchs first for the `MGLS_LICENSE_FILE` environment variable. This may be useful to avoid license conflits with other FPGA software, like Lattice Diamond, sinc it uses the same environment variable (`LM_LICENSE_FILE`).

```bash
> python.exe .\run.py                                                                                                                    
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd                        passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\path\src\path.vhd                                    passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\print_pkg.vhd                            passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\types.vhd                             passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec_builder.vhd                     passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec_builder-2008p.vhd               passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec.vhd                             passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\codec-2008p.vhd                       passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_string_pkg.vhd           passed           
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_string_pkg.vhd(40): Function 'read_char' may complete wi
thout a RETURN at line 39.                                                                                                               
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_string_pkg.vhd(46): Function 'get_ptr' may complete with
out a RETURN at line 45.                                                                                                                 
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\string_ptr_pkg.vhd                    passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\ansi_pkg.vhd                             passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_levels_pkg.vhd                       passed           
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_levels_pkg.vhd(33): (vcom-1246) Range 1 to 0 is null.               
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\string_ptr_pkg-body-2002p.vhd         passed           
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\string_ptr_pkg-body-2002p.vhd(289): Function 'to_string' may complete
 without a RETURN at line 287.                                                                                                           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\byte_vector_ptr_pkg.vhd               passed 
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_integer_vector_pkg.vhd   passed           
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_integer_vector_pkg.vhd(40): Function 'read_integer' may 
complete without a RETURN at line 39.                                                                                                    
** Warning: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\api\external_integer_vector_pkg.vhd(46): Function 'get_ptr' may compl
ete without a RETURN at line 45.                                                                                                         
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\integer_vector_ptr_pkg.vhd            passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_handler_pkg.vhd                      passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\print_pkg-body.vhd                       passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\logger_pkg.vhd                           passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\dictionary\src\dictionary.vhd                        passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\run\src\run_types.vhd                                passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\file_pkg.vhd                             passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\logging\src\log_handler_pkg-body.vhd                 passed           
Compiling into vunit_lib: C:\Python39\Lib\site-packages\vunit\vhdl\data_types\src\integer_vector_ptr_pkg-body-2002p.vhd passed           
[... cut ...]
==============================================                                                                                           
Total time was 2.6 seconds                                                                                                               
Elapsed time was 2.7 seconds                                                                                                             
==============================================                                                                                           
All passed!
```

closed 2021-03-23 by rafaelnp