VUnit/vunit #715: Recompile RTL with GUI open

vperrin59, 2021-04-29:
I've been facing the following issue and I don't know if there is a limitation or there is another way to do it.

First I'm launching a test interactively with --gui option. The simulator pops up.

Then I want to recompile the code after some modifications I've done keeping the GUI open.

run.py --compile.

It will not compile because of some lock file handled by the GUI I suppose. Is there a way to overcome this ?

I'm using xcellium and verilog/systemverilog

vperrin59, 2021-05-07:
I found a solution that worked on my side, in case anybody else is interested:

I just add to remove option '-nclibdirname "%s"' % (str(Path(self._output_path) /  "libraries")) in simulate method in incisive.py

abyszuk, 2021-06-11:
At least in Modelsim/Questa you can type `vunit_restart` in the TCL console of the simulator GUI and it will also recompile files if needed.

johonkanen, 2021-11-24:
At least with windows + ghdl + gtkwave combination the issue is that vunit deletes the test out folder so opening gtkwave locks the folder where the ghdl waveform file .ghw file is thus making the test fail. The simple solution to this is just to call gtkwave from outside the test_out folder in a separate terminal. For example  run the uart tests from vunit examples with 24 threads
> python run.py -p 24 --gtkwave-fmt ghw

and then open gtkwave in another terminal with
>  gtkwave .\vunit_out\test_output\tb_uart_lib.tb_uart_rx.test_receives_one_bytee4df12afe1e550e414a1b5c4e140c76e186f819f\ghdl\wave.ghw

then just ctrl-r away (: