VUnit/vunit #725: Parsing problem in 4.4.0?

mschiller302, 2021-07-26:
I have two source files that represent two versions of a block:
vita_49_2_sink_buffer_output_logic_read_2_0.vhd
vita_49_2_sink_buffer_output_logic_read_1_0.vhd

As the name suggests they represent two versions of the same block in my vunit project.

I get the following warning:

WARNING -  /...../vita_49_2_sink_buffer_output_logic_read_1_0.vhd: context 'checking' previously defined in /...../vita_49_2_sink_buffer_output_logic_read_2_0.vhd

The offending lines of code appear to be in these generate sections (I can't attach the full VHDL file so a snipit will have to do)

On the right is the original version which appears verbatim in both 2_0 and 1_0 of this block.   But if I make the changes indicated on the left, the warning goes away
[image]

This appears to be a parsing bug.

LarsAsplund, 2021-07-26:
This is our simple regex-based parser that picks up the report string

Context Checking is

as the beginning of a context statement. It is not a proper parser that understands that this is part of a string. You can make changes to how you formulate the string or simply ignore the warning. We may change to a proper parser in the future but we're hoping for some of the emerging open source parser to do the work for us rather than we spending time on that.

eine, 2021-07-29:
@mschiller302 it would be interesting if you dropped your sources into pyVHDLModel (https://github.com/vhdl/pyVHDLModel#list-all-entities-with-generics-and-ports) for testing. That is a VHDL language model which uses GHDL for parsing and analysing the sources. There is a GUI proof of concept in https://umarcor.github.io/osvb/apis/project.html#open-source-vhdl-design-explorer-osvde.

pyVHDLModel is a possible replacement for VUnit's parser. However, since it needs GHDL at the moment, it is unlikely to be used as the single and default solution. In the future, [pyVHDLParser](https://github.com/Paebbels/pyVHDLParser/) might be used instead of pyGHDL.dom.

LarsAsplund, 2024-12-11:
see #1080

closed 2024-12-11 by LarsAsplund