VUnit/vunit #727: CHECK_EQUAL_VARIANCE with variance of 0 should default to CHECK_EQUAL

jonatasp1016, 2021-08-16:
The CHECK_EQUAL_VARIANCE macro in the vunit_defines.svh file does the following assertion.

`
define CHECK_EQUAL_VARIANCE(got,expected,variance,msg=__none__) 
         assert (((got) < ((expected) + (variance))) && ((got) > ((expected) - (variance)))) else
....
`

I tried using the macro by passing the variance as a percentage of a variable: `CHECK_EQUAL_VARIANCE(got.bucket_a, result.bucket_a, result.bucket_a*0.002)` , but for cases in which result.bucket_a was zero, the check failed, basically because the variance passed was also zero. I believe it would be more correct for the macro to check with less than or equal to / greater than or equal to instead of plain less than / greater than. 

I.e.:
(((got) <= ((expected) + (variance))) && ((got) >= ((expected) - (variance)))) 

instead of:
(((got) < ((expected) + (variance))) && ((got) > ((expected) - (variance))))

In that case, CHECK_EQUAL_VARIANCE would implicitly default to CHECK_EQUAL in case of variance=0

LarsAsplund, 2021-08-18:
@dbalthazor What is you view on this?  I know that at least Python `unittest` accept values right on the boundaries of the range.

dbalthazor, 2021-08-18:
I don't have particularly strong feelings one way or another, I'll let you make the call whether that comparison should be inclusive or exclusive.  Either way, a test case for variance 0 should be added that checks for this edge case.

LarsAsplund, 2021-08-22:
@jonatasp1016 @dbalthazor Then I suggest we change. A PR @jonatasp1016?