VUnit/vunit #732: VUnit does not look in all libraries when compiling Xilinx primitives

anro7, 2021-08-31:
Hello,

I am trying to simulate a larger top level design (entire Series-7 FPGA), featuring several transceiver-based IPs. So far, I have been able to simulate this design when not using VUnit (just my normal simulator, which btw is ActiveHDL 11.1). In VUnit however, I run into the following error at elaboration:

```
ELBREAD: Elaboration process.
ELBREAD: Error: ELBREAD_0081 GTPE2_CHANNEL.v (3315): Design unit B_GTPE2_CHANNEL instantiated in simprims_ver.GTPE2_CHANNEL not found in searched libraries: simprims_ver, lib.
ELBREAD: Error: Elaboration process completed with errors.
VSIM: Error: Simulation initialization failed.
```

From what I have been reading on Xilinx forums, the B_GTPE2_CHANNEL is a primitive originating from the secureip library from Xilinx. In my python script, I have added all the precompiled Vivado libraries as external libraries and also added them as arguments to the vlog command, as follows:

```
vu.add_external_library("secureip",     "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/secureip")
vu.add_external_library("simprims_ver", "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/simprims_ver")
vu.add_external_library("unimacro",     "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/unimacro")
vu.add_external_library("unimacro_ver", "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/unimacro_ver")
vu.add_external_library("unisim",       "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/unisim")
vu.add_external_library("unisims_ver",  "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/unisims_ver")
vu.add_external_library("xpm",          "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/xpm")
vu.add_external_library("xilinx_vip",   "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/xilinx_vip")
vu.add_external_library("aldec",        "C:/Aldec/Active-HDL-11.1/vlib")

vu.add_external_library("fifo_generator_v13_2_5",         "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/fifo_generator_v13_2_5")
vu.add_external_library("blk_mem_gen_v8_4_4",             "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/blk_mem_gen_v8_4_4")
vu.add_external_library("generic_baseblocks_v2_1_0",      "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/generic_baseblocks_v2_1_0")
vu.add_external_library("axi_data_fifo_v2_1_19",          "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_data_fifo_v2_1_19")
vu.add_external_library("axi_infrastructure_v1_1_0",      "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_infrastructure_v1_1_0")
vu.add_external_library("axi_register_slice_v2_1_20",     "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_register_slice_v2_1_20")
vu.add_external_library("axi_crossbar_v2_1_21",           "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_crossbar_v2_1_21")
vu.add_external_library("axi_protocol_converter_v2_1_20", "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_protocol_converter_v2_1_20")
vu.add_external_library("util_vector_logic_v2_0_1",       "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/util_vector_logic_v2_0_1")
vu.add_external_library("axi_clock_converter_v2_1_19",    "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_clock_converter_v2_1_19")
vu.add_external_library("axi_dwidth_converter_v2_1_20",   "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_dwidth_converter_v2_1_20")
vu.add_external_library("lib_pkg_v1_0_2",                 "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/lib_pkg_v1_0_2")
vu.add_external_library("lib_fifo_v1_0_14",               "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/lib_fifo_v1_0_14")
vu.add_external_library("lib_bmg_v1_0_13",                "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/lib_bmg_v1_0_13")
vu.add_external_library("axi_pcie_v2_9_2",                "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_pcie_v2_9_2")
vu.add_external_library("axi_mmu_v2_1_18",                "C:/Aldec/Active-HDL-11.1/vlib/xilinx_vivado/axi_mmu_v2_1_18")

lib = vu.add_library("lib")

vu.add_compile_option("activehdl.vlog_flags", ["-dbg", "-work work", "-v2k5", "-l secureip", "-l simprims_ver", "-l unimacro", "-l unimacro_ver", "-l unisim", "-l unisims_ver", "-l xpm", "-l xilinx_vip", "-l work", "-l axi_register_slice_v2_1_20", "-l fifo_generator_v13_2_5", "-l axi_mmu_v2_1_18", "-l util_vector_logic_v2_0_1", "-l blk_mem_gen_v8_4_4", "-l lib_pkg_v1_0_2", "-l lib_fifo_v1_0_14", "-l lib_bmg_v1_0_13", "-l axi_pcie_v2_9_2", "-l axi_data_fifo_v2_1_19", "-l generic_baseblocks_v2_1_0", "-l axi_crossbar_v2_1_21", "-l axi_protocol_converter_v2_1_20", "-l axi_dwidth_converter_v2_1_20", "-l axi_infrastructure_v1_1_0", "-l axi_clock_converter_v2_1_19", "DxiV3ModuleTop_Tb"])

vu.add_compile_option("activehdl.vcom_flags", ["-dbg", "-2008", "-relax"])
```

 
To me, this seems like a VUnit problem, specifically that at compilation/elaboration it does not look in all of the available libraries, but only in **simprims_ver** and **lib**. Is there any way for me to get around this and force VUnit to look in all external libraries? (I thought it would be sufficient if I add the -l arguments to the vlog command pointing to all the external libraries, but apparently not).

I would be grateful for any suggestions. Thank you.

anro7, 2021-08-31:
Found my solution in the meantime. Just needed to add some extra simulation options.

vu.set_sim_option("activehdl.vsim_flags", [**"-L unisim", "-L work", "-L secureip"**, "+access", "+r"])

With the extra -L directives, it now looks in all the required libraries.

LarsAsplund, 2021-09-02:
Could you do a clean compile (`--clean`) and also add ` --log-level=debug`. Then provide the full output,

Also, waht version are you using?