VUnit/vunit #733: Support for simulating precompiled artifacts

solsjo, 2021-08-31:
VUnit does a great job of deriving the dependency tree for the compile phase. This relieves the developer of the burden of maintaining a buildsystem. 

But there are benefits with allowing a buildsystem to relieve VUnit of that work as well.
For instance,
the build system bazel + surrounding tools, enables artifacts to be built and cached and tested remotely.
The benefit being that if an artifact already exists remotely it won't have to be rebuilt locally.
If rebuilt the (ghdl linking) or update of the lib / library.cfg file is only a small increment and a fast operation.
If a bazel test-rule depends on an artifact that hasn't changed then the test won't need to run, conversely only a tests with a dependency change needs to be tested. This reduces the amount of tests needed to run during e.g. a commit hook or CI pipeline for a pull request.

For this to work VUnit cannot be allowed to know the whole dependency tree. Rather it only needs the
- test-bench file to parse
- the python test configuration file
- the test artifact

In the extension I could write bazel rules for ghdl compilation and vunit tests.

I have prepared a [commit](https://github.com/solsjo/vunit/commit/8026e25c6c494e0ed8cc82049ac9085970c2a511) which describes the intension, but I'm sure there are more correct ways of handling it.

I'd be happy to contribute and can make the pull request myself if I get feedback on how it should be added.
I'm also perfectly fine with someone else doing it, the code change in my commit has pretty small. (test are missing though, as this is still just at the idea stage)

umarcor, 2021-09-06:
Hi @solsjo! That sounds interesting! Are you ware of [hdl/bazel_rules_hdl](https://github.com/hdl/bazel_rules_hdl)? The purpose of that repository is gathering Bazel recipes for HDL|EDA related tools. It would be interesting to gather rules for GHDL compilation and for running VUnit in there.

Other than that, I really like the idea, specially for sharing test results of large open source projects. However, I'm not sure about VUnit providing enough info for being able to reuse test artifacts. In order to be sure about pre-compiled artifacts being reusable, two conditions need to be met:

- The artifacts themselves are readable on different machines. With GHDL, the `.cf` might be reusable, but `.o` objects (generated by LLVM and GCC backends) are not portable. Therefore, artifacts need to be identified by the compiler/platform, apart from VUnit's own parameters.
- VUnit provides enough info about the project, the parameters and the tests to uniquely identify a "case". I think this is possible, just maybe challenging to implement.

solsjo, 2021-10-01:
@umarcor 
I haven't looked into it that much but I've read forum posts about perserving the bazel cache online in https://github.com/gitpod-io/gitpod

But I'm not certain that that would work in-between users.
There are most probably other alternatives as well.

nathanaelhuffman, 2023-12-09:
@solsjo I'm interested in building some vunit rules for buck2 and figured taking a look at these would be helpful, but the https://github.com/solsjo/rules_vunit seems to be gone now? Is there a copy that is accessible somewhere?

solsjo, 2024-01-12:
Hi @nathanaelhuffman!

Missed the email notification, sorry. I made it private a while back, but can make it public again.

From what remember I might have tied it up somewhat to some ghdl rules that I also wrote. Those rules however, are somewhat convoluted as I at the time was new to the build chain for hdl code. Still, some way of building the vunit hdl code using a build system might be a good idea even if VUnit's compilation is very performant. 

I'll let you know when it has been made public again.