VUnit/vunit #740: Question: Is there a way for Vunit to only rerun previously failed testcases?

philipphct, 2021-09-10:
Hi everyone,

I was wondering: 
Lets say I got quite a handfull of testcases, some of them may have failed in a previous simulation run (across all testbeches in my run.py), is there a builtin way for Vunit to determine what testcases failed previously, to only run those in next time I execute my projects run.py?

If not, would it be a viable option to write the results in a post_run function to a file and somehow overwrite the test selection?

umarcor, 2021-09-10:
> If not, would it be a viable option to write the results in a post_run function to a file and somehow overwrite the test selection?

Yes. You can do this in the `run.py` file. On the one hand, as you said you can use post_run to save the tests you want to run into a file. On the other hand, you can look for that file, and override the CLI arguments for specifying the tests.

Let's say we are dealing with https://github.com/VUnit/tdd-intro/tree/master/tutorial/exercise_02/solution:

```sh
# python3 run.py -l
tb_lib.tb_counter.all
tb_lib.tb_counter_with_test_cases.Test counting
tb_lib.tb_counter_with_test_cases.Test that data can be loaded
tb_lib.tb_counter_with_test_cases.Test reset
Listed 4 tests
```

Add the example from http://vunit.github.io/py/vunit.html?highlight=post_func#vunit.ui.results.TestResult:

```sh
# python3 run.py
...
T:/vunit/tdd-intro/tutorial/exercise_02/solution/vunit_out
tb_lib.tb_counter.all
passed
0.6610262393951416
T:/vunit/tdd-intro/tutorial/exercise_02/solution/vunit_out/test_output/tb_lib.tb_counter.all_76b52425acb60f6915a36263a5616ff280e3a7a8
tb_lib.tb_counter.all_76b52425acb60f6915a36263a5616ff280e3a7a8
tb_lib.tb_counter_with_test_cases.Test counting
passed
0.6062884330749512
T:/vunit/tdd-intro/tutorial/exercise_02/solution/vunit_out/test_output/tb_lib.tb_counter_with_test_cases.Test_counting_075666daafc659064e4fb1bfdbef1dbb5a6fa2f2
tb_lib.tb_counter_with_test_cases.Test_counting_075666daafc659064e4fb1bfdbef1dbb5a6fa2f2
tb_lib.tb_counter_with_test_cases.Test that data can be loaded
passed
0.6314296722412109
T:/vunit/tdd-intro/tutorial/exercise_02/solution/vunit_out/test_output/tb_lib.tb_counter_with_test_cases.Test_that_data_ceeb64a53e8c0727429308beeeb2921ef8a8a0fe3
tb_lib.tb_counter_with_test_cases.Test_that_data_ceeb64a53e8c0727429308beeeb2921ef8a8a0fe3
tb_lib.tb_counter_with_test_cases.Test reset
passed
0.6607403755187988
T:/vunit/tdd-intro/tutorial/exercise_02/solution/vunit_out/test_output/tb_lib.tb_counter_with_test_cases.Test_reset_0d305aff74541bf3125ce21ad148a1e9e41741e0  
tb_lib.tb_counter_with_test_cases.Test_reset_0d305aff74541bf3125ce21ad148a1e9e41741e0
```

So, the `key` is what you want to save. As those are the ones you can use in CLI:

```sh
# python3 run.py tb_lib.tb_counter_with_test_cases.'Test counting' tb_lib.tb_counter_with_test_cases.'Test reset'
Re-compile not needed
...
```

Actually, VUnit saves tests-to-results-directory mappings in a file:

```sh
# cat vunit_out/test_output/test_name_to_path_mapping.txt
tb_lib.tb_counter_with_test_cases.Test_counting_075666daafc659064e4fb1bfdbef1dbb5a6fa2f2 tb_lib.tb_counter_with_test_cases.Test counting
tb_lib.tb_counter_with_test_cases.Test_reset_0d305aff74541bf3125ce21ad148a1e9e41741e0 tb_lib.tb_counter_with_test_cases.Test reset
tb_lib.tb_counter_with_test_cases.Test_that_data_ceeb64a53e8c0727429308beeeb2921ef8a8a0fe3 tb_lib.tb_counter_with_test_cases.Test that data can be loaded
```

It would make sense to use `vunit_out/test_output/failed_tests.txt` as the file to save to in the post_run hook.

Then, the way to override/extend the CLI arguments is providing a custom list of strings (http://vunit.github.io/py/vunit.html?highlight=post_func#vunit.ui.VUnit.from_argv):

```py
from sys import argv
from vunit import VUnit

PRJ = VUnit.from_argv(argv + ['my', 'list', 'of', 'test', 'keys'])
...
```

Say:

```py
from sys import argv
from pathlib import Path
from vunit import VUnit

ROOT = Path(__file__).parent

failed_tests = ROOT / 'vunit_out/test_output/failed_tests.txt'
if failed_tests.exists():
  with failed_tests.open('r') as fptr:
      PRJ = VUnit.from_argv(argv + fptr.readlines())
else
  PRJ = VUnit.from_argv()
...
```

This is a naive approach, using a text file to save each key in a row. You might use JSON, YAML or XML instead. VUnit does have built-in support for generating an XUnit report file. Therefore, you might avoid writing a file yourself, and instead read the XML generated by VUnit (which does include the result of each tests). That might be more elegant, but it also involves some more effort when parsing/reading the XML.

philipphct, 2021-09-10:
Thanks! Works like a charm :)

closed 2021-09-10 by philipphct

umarcor, 2021-09-10:
That is nice! It's a cleaner solution indeed. However, I did not know how experienced you are with VUnit/Python.

BTW, if you consider this to be a valuable feature, we can evaluate maybe generating the failure list file or providing a utility function/option to filter the tests. I provided a ready to use solution, but there is room for improvement should you want to pick the challenge anytime in the future.