VUnit/vunit #741: Enter GUI upon failed test?

sibeov, 2021-09-16:
Hi.

Wondered if it is possible to enter GUI upon the first fail in a test. Is it possible to "catch" the fail someplace and rerun the previous test with the -g flag? Or any other similar "workaround" possible? Would be a nice feature. Was not able to find any information or anything else indicating such a feature exists. If so please redirect me.

Kind regards.

eine, 2021-09-16:
See #740.

sibeov, 2021-09-16:
@eine Ah! Thanks. Looks like what I am looking for! I will close issue when I can confirm it works as I wish.

dalex78, 2022-05-12:
@sibeov did it resolve your issue ?

sibeov, 2022-05-19:
Hi @dalex78 !

Sorry for late reply. I have not had the time to check yet because time did not allow me to
go further into the matter. I did get some pointers and tried to create my own XML parsing tool and 
implement that into a run.py script without any luck. If time permits one day I will retry this.

So Sorry for the late response on my part. 
You may close this issue for now. I can reopen a new issue or similar. If it is no problem
this issue can remain open.

Kind regards

LarsAsplund, 2022-05-29:
@sibeov Since you're not looking into re-running a set of failing tests as outlined in #740 you can simply do

``` console
python run.py --fail-fast
```

It stops on the first failing test. If that test is `lib.tb_example.failing_test` you simply follow up with

```console
python run.py lib.tb_example.failing_test` -g
```

sibeov, 2022-07-14:
@LarsAsplund 

Do you mean then to write a script which checks which testcase fails and then run that with the `-g` flag?
Thanks for the tip, it might work. I will see if that works out for me next time I need this. Thanks!