VUnit/vunit #752: WARNING - failed to find a primary design unit 'scoreboardpkg_int' in library 'osvvm'

rafaelnp, 2021-10-05:
When using the following setup:

Linux:
```Linux archlinux 5.10.64-1-lts #1 SMP Sun, 12 Sep 2021 09:11:21 +0000 x86_64 GNU/Linux```

Python: 
```
python --version
Python 3.9.7
```

VUnit (installed using pip): 
```
pip show vunit_hdl
Name: vunit-hdl
Version: 4.5.0
```

GHDL (archlinux package):

```
ghdl -v
GHDL 1.0.0 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 11.1.0
 GCC back-end code generator
Written by Tristan Gingold.
```

I am experiencing the following warning when running VUnit + OSVVM (that leads to an error): 

```
❯ python run.py -l
lib.test_tb.all
Listed 1 tests
❯ python run.py
WARNING - /home/rnp/src/fpga/vhdl/misc/test_tb/test_tb.vhd: failed to find a primary design unit 'scoreboardpkg_int' in library 'osvvm'
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/verification_components/src/axi_pkg.vhd                     passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/string_ops/src/string_ops.vhd                               passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/path/src/path.vhd                                           passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/VendorCovApiPkg.vhd                                   passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/TranscriptPkg.vhd                                     passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/TextUtilPkg.vhd                                       passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/NamePkg.vhd                                           passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/OsvvmGlobalPkg.vhd                                    passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/AlertLogPkg.vhd                                       passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/SortListPkg_int.vhd                                   passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/ResolutionPkg.vhd                                     passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/TbUtilPkg.vhd                                         passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/RandomBasePkg.vhd                                     passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/RandomPkg.vhd                                         passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/MessagePkg.vhd                                        passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/MemoryPkg.vhd                                         passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/CoveragePkg.vhd                                       passed
Compiling into osvvm:     ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/osvvm/OsvvmContext.vhd                                      passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/logging/src/print_pkg.vhd                                   passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/types.vhd                                    passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/codec_builder.vhd                            passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/codec_builder-2008p.vhd                      passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/codec.vhd                                    passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/codec-2008p.vhd                              passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/api/external_string_pkg.vhd                  passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg.vhd                           passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/logging/src/ansi_pkg.vhd                                    passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/logging/src/log_levels_pkg.vhd                              passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd                passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/byte_vector_ptr_pkg.vhd                      passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/api/external_integer_vector_pkg.vhd          passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/data_types/src/integer_vector_ptr_pkg.vhd                   passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/logging/src/log_handler_pkg.vhd                             passed
Compiling into vunit_lib: ../../../../../.local/lib/python3.9/site-packages/vunit/vhdl/logging/src/print_pkg-body.vhd                              passed
[... 109 lines cut ...]
	shared variable scoreboard : ScoreboardPType;
	shared variable coverage   : CovPType;

begin
	process begin
		test_runner_setup(runner, runner_cfg);
		wait for 100 ns;
		test_runner_cleanup(runner);
	end process;
end architecture tb;
```

and the `run.py` to run it:
```python
from vunit import VUnit

VU = VUnit.from_argv()
VU.add_osvvm()
VU.add_verification_components()
VU.set_compile_option('ghdl.a_flags', ['-Wno-hide'])
VU.add_library("lib").add_source_files(["*.vhd"])
VU.main()
```

When using ModelSim Intel FPGA Starter Edition 2020.1, this problem does not show up. I suppose I am missig some configuration or minor detail regarding GHDL. Thanks in advance. :)

umarcor, 2021-10-06:
- https://github.com/VUnit/vunit/blob/master/vunit/builtins.py#L168-L176
- https://github.com/VUnit/vunit/blob/master/vunit/sim_if/__init__.py#L153-L157
- https://github.com/VUnit/vunit/blob/master/vunit/sim_if/modelsim.py#L75-L79

I think the solution is to override `supports_vhdl_package_generics` in the GHDL sim_if: https://github.com/VUnit/vunit/blob/6114f044401d7fe94098ed5bbfb8f6d575dcfcac/vunit/sim_if/ghdl.py. Currently, GHDL is using the default `False`.

Hence, this is a legit bug to be fixed in VUnit.

rafaelnp, 2021-10-07:
@umarcor I tested here your changes and it worked. Thanks for your reply and pull resquest. 👍