VUnit/vunit #782: Vivado compile order scanning fails with .coe or .mif dependencies

smenzel, 2021-12-06:
vivado.py function _read_compile_order() used by add_vivado_ip() fails with assertion when compile_order.txt contains references to files that are not VHDL or Verilog. However, some Vivado IP include .coe or .mif memory initialisation files. 

Ideally, these ressource files should get copied (or symlinked) by VUnit to the simulator_output_path, so that they are available for the simulator. At the least, the assertion failure at line vivado.py:93 should  be reduced to a warning (and the line ignored), so that the function does not fail.

My dirty workaround right now is changing the assert to a continue and using a pre_config to copy the ressource later.

vivado.py:93
```
if file_type not in ("Verilog", "VHDL", "Verilog Header")
    print("Warning: compile_order.txt entry ignored: %s" % line)
    continue
```

run.py
```
def pre_config (output_path, simulator_output_path):
    shutil.copy(root + "hdl/ip/lutram/lutram.mif", simulator_output_path)
    return True

for tb in lib.get_test_benches():
    tb.set_pre_config(pre_config)
```

LarsAsplund, 2021-12-07:
Since this function has to do with compile order I don't think it should do smartness with other types of files so I suggest that there is a new argument `fail_on_non_hdl_files` with a default value `True` to keep backwards compatibility. Want to do a PR?

smenzel, 2021-12-08:
I have to see how I can make a PR from my work computer. In the meantime, here is a patch:
```
diff --git a/vunit/vivado/vivado.py b/vunit/vivado/vivado.py
index e36da72a..d14dd56b 100644
--- a/vunit/vivado/vivado.py
+++ b/vunit/vivado/vivado.py
@@ -13,11 +13,12 @@ from os import makedirs
 from pathlib import Path
 
 
-def add_from_compile_order_file(vunit_obj, compile_order_file, dependency_scan_defaultlib=True):
+def add_from_compile_order_file(vunit_obj, compile_order_file, dependency_scan_defaultlib=True,
+                                fail_on_non_hdl_files=True):
     """
     Add Vivado IP:s from a compile order file
     """
-    compile_order, libraries, include_dirs = _read_compile_order(compile_order_file)
+    compile_order, libraries, include_dirs = _read_compile_order(compile_order_file, fail_on_non_hdl_files)
 
     # Create libraries
     for library_name in libraries:
@@ -77,7 +78,7 @@ def create_compile_order_file(project_file, compile_order_file, vivado_path=None
     )
 
 
-def _read_compile_order(file_name):
+def _read_compile_order(file_name, fail_on_non_hdl_files):
     """
     Read the compile order file and filter out duplicate files
     """
@@ -90,7 +91,14 @@ def _read_compile_order(file_name):
 
         for line in ifile.readlines():
             library_name, file_type, file_name = line.strip().split(",", 2)
-            assert file_type in ("Verilog", "VHDL", "Verilog Header")
+
+            if not  file_type in ("Verilog", "VHDL", "Verilog Header"):
+                if fail_on_non_hdl_files:
+                    raise RuntimeError("Unsupported compile order file: %s" % file_name)
+                else:
+                    print("Compile order file ignored: %s" % file_name)
+                    continue
+
             libraries.add(library_name)
 
             # Vivado generates duplicate files for different IP:s
```

closed 2022-11-07 by LarsAsplund