VUnit/vunit #787: Possible to include/compile MicroSemi IPs?

sibeov, 2022-01-03:
Hello.

Anyone has any experience or have succeeded in compiling MicroSemi IPs using VUnit?
I know it is not supported out of the box. If this is not the right place for questions then please
redirect me.

Kind Regards

nfrancque, 2022-01-12:
@sibeov There was some gitter discussion about this [here](https://gitter.im/VUnit/vunit?at=60f13fc1d2556414f568eec1), you can ask more there if needed.  In general, you can do whatever you are already doing for the microsemi IP, and then inform vunit that the compile artifacts are there.

LarsAsplund, 2022-01-13:
@sibeov I also recommend the gitter channel as the better forum for such a discussion. There you can find people with such experiences.

closed 2022-01-13 by LarsAsplund