VUnit/vunit #793: Better support for switching simulators

nfrancque, 2022-01-12:
I have a vunit fork [here](https://github.com/nfrancque/vunit) with some changes for our company's infrastructure that I'd like to get merged in where possible, this is the first one.

Would like hear from others what the best way to handle this is before making a PR, might not be the way I have done so far.

We have a mixed simulator environment so going back and forth without impacting the other simulators is important.  The environment variable is a little too permanent, the argument is nice to have.

At a high level, it needs

1) Command line argument to select simulator, in addition to environment variable.  
- In my fork, I just expose the option and set the environment variable accordingly.  Obviously this wouldn't be the real solution.

2) Differentiate between different versions and variants of tools. For instance, "modelsim" includes several tiers of modelsim, Intel's baked in modelsim, questasim, etc.  Other simulators may have different versions installed. 
- For this one I have only differentiated between intel modelsim, all tiers/versions of Mentor modelsim, and all tiers/versions of Questasim.  Multiple versions of each are not supported, but you can switch between these three without breaking the others.
- This seems like the hardest part as the number of potential simulators is unknown.

3) If using Xilinx IP, the artifacts for this also need to be separated and not interfere with each other.  
- To do this I've just changed the vivado compilation to happen in the respective output folders.  It's in a private repo not in the fork, but based on [this example](https://github.com/VUnit/vunit/tree/master/examples/vhdl/vivado/).  I can share if needed.

4) More of a nice to have, a `--clean` that optionally takes a simulator interface and only clears that simulators output artifacts.  Just a time saver.  I have not done this.

I haven't done anything to separate out the output[_with_color].txt but this could also be something to consider.

Currently this relies on each tool being installed in a pre-configured list of valid install paths.  This is not the right way to do it globally, but makes it easier for us.  I'd be ok with removing this and just having it in the private half of our repo.  Open to any suggestions.

@LarsAsplund Any thoughts?  @GlenNicholls I think I've seen you say before that you also use multiple simulators

GlenNicholls, 2022-01-12:
Yeah, I can provide some input. I found it difficult to deal with some of these things, but I think that's mostly because I wasn't familiar with VUnit when I started that endeavor. Once I got something working, I didn't look any further to find a better way to integrate it. @cmarqu is probably good to have on here as well, I seem to remember him sharing snippets with me a long time ago for similar things.

>Command line argument to select simulator, in addition to environment variable.

How exactly would this look if it were to be implemented in VUnit? Would the user create multiple env vars for each tool/version and VUnit would parse all variables that match some pattern to populate this option for what is available? We don't use a huge array of tool versions because we don't have anything safety-critical so our variables just point to whatever is latest for each tool.

I definitely see value in changing the simulator with a flag... I've been using aliases and those get pretty annoying after a while because sometimes you aren't sure which simulator you ran with if you're not watching the logs. 

>Differentiate between different versions and variants of tools. For instance, "modelsim" includes several tiers of modelsim, Intel's baked in modelsim, questasim, etc. Other simulators may have different versions installed.

How do you differentiate or know what's available? Do you just install things in the default/predefined location and parse? This is how I handled pre-generating output products that are out of date before running VUnit. If the user doesn't specify which tool with a variable or args, I search default locations and prompt the user to decide. If nothing is there, it just raises an exception. I don't really like this, though, because it just adds another layer on top of VUnit, but I haven't thought of a better way to do it.

>If using Xilinx IP, the artifacts for this also need to be separated and not interfere with each other.

This slightly confuses me, but I think I understand what you mean. Are you saying that when "compiling" an IP for modelsim vX you compile to modelsim_vX dir instead of just modelsim to avoid interfering with modelsim vY? 

>More of a nice to have, a --clean that optionally takes a simulator interface and only clears that simulators output artifacts. Just a time saver. I have not done this.

I agree, this would be really nice to have... I think a bare `--clean/-c` should still remove everything, but `--clean=` might be nice to specify what specifically to remove.

>Currently this relies on each tool being installed in a pre-configured list of valid install paths. This is not the right way to do it globally, but makes it easier for us. I'd be ok with removing this and just having it in the private half of our repo. Open to any suggestions.

What other options are there (for multiple versions of tools)? The only things that come to mind are the following:

* Aliases/scripts for setting env vars -> no change required to VUnit
* Setting/overriding a parameter that sets the pattern/regex defining where tools are installed
* Some reserved pattern for environment variables 

One thing I think this issue highlights is that VUnit isn't very easy to integrate in large environments, at least that's been my experience. When things are confined to a single repo where you can simply glob for files it's perfect. However, as soon as you have to start adding hooks to find files, configure different simulators consistently between repos, deal with IP, and work with internal HDL tools it gets pretty hairy. I simply wrapped VUnit in another class to satisfy the interface I need, but that's been slowly killing me inside because it's become more maintenance than it's worth.

nfrancque, 2022-01-13:
> How do you differentiate or know what's available? 

Good point, it might be better to expose options like these as a member function of `vu` and let individuals elevate it to their own command line argument if needed.  The not being sure what simulator is running was a problem for us too.  I have a bunch of infrastructure on top of vunit to manage stuff like this, changes in the tb init / teardown vhdl functions that get exposed as an argument, etc.

> Are you saying that when "compiling" an IP for modelsim vX you compile to modelsim_vX dir instead of just modelsim to avoid interfering with modelsim vY?

Pretty much.  Just make sure that any output artifacts (actual compile, Vivado XPM, Vivado IP, whatever) is contained within the simulator directory.

> --clean= might be nice

This is what I was thinking too, destroy everything by default but provide the ability to only do a particular simulator.

> One thing I think this issue highlights is that VUnit isn't very easy to integrate in large environments

Agree.  Maybe someday I can open source the infrastructure around vunit because like you I'm the "vunit guy" and have a bunch of stuff to make this easy to use at enterprise scale.  Would hate to have to redo this.  Files are declared in manifests that interface with vunit, vsg, etc and declare what simulators are able to run these tests.  Some improved reporting of test results since we spawn something like a few hundred to a few thousand tests per block to permute a bunch of reasonable generic combinations.  So there is code to generate a csv of this to see what conditions it fails under.  Lots of extra crap to do.

@eine was talking about this a while ago too.

Here's an example of what our run.py's look like for a block, everything other than this gets handled behind the scenes calling `add_test_configs` for each block.

```python

from pathlib import Path
from vunit_helper import run_vunit_main_default, gen_configs, add_config

def add_test_configs(prj, tb):
    test_cfg_options = {
        "tdata_width": [8, 16, 32, 64],
        "master_stall": [0.0, 0.1, 0.3, 0.5, 0.7, 0.9],
        "slave_stall": [0.0, 0.1, 0.3, 0.5, 0.7, 0.9],
    }

    configs = gen_configs(test_cfg_options)

    for test in tb.get_tests():
        for tmp_config in configs:
            config = tmp_config.copy()
            add_config(config, test)

    return prj

```

tasgomes, 2022-01-13:
Hi @nfrancque,

I have dealt with some of your issues without changing VUnit source files and only using per-existing VUnit functionalities.

> 1. Command line argument to select simulator, in addition to environment variable.

I added 2 custom command line arguments (simulator name and simulator path) to the `run.py`. E.g.:
```
cli = VUnitCLI()
cli.parser.add_argument(
        "--simulator",
        help="Define the name of simularo",
    )
```
and then I assigned them inside:
```
environ["VUNIT_SIMULATOR"] = args.simulator
environ["VUNIT_MODELSIM_PATH"] = args.simulator_path
environ["VUNIT_GHDL_PATH"] = args.simulator_path
```
> If using Xilinx IP, the artifacts for this also need to be separated and not interfere with each other.

I have done the same, I compile the Vivado libs in different folders (depending on the simulator) but again I used a custom command line argument to provide the location of the Vivado lib, which is different depending on the simulator used.

> Differentiate between different versions and variants of tools. For instance, "modelsim" includes several tiers of modelsim, Intel's baked in modelsim, questasim, etc. Other simulators may have different versions installed.

This could be done with adding more information to the corresponding folder, e.g.:
```
/sim_lib/modelsim_pe/unisim
/sim_lib/modelsim_ae/unisim
/sim_lib/ghdl/unisim
```
> More of a nice to have, a --clean that optionally takes a simulator interface and only clears that simulators output artifacts. Just a time saver. I have not done this.

For this, what I do is to define the VUnit output folder with a different names using the existent "-o" command line argument. E.g.
```
run.py -o /vunit_out/modelsim_pe
run.py -o /vunit_out/modelsim_ae
run.py -o /vunit_out/ghdl
```
This makes sure that each simulator runs on its output folder so `--clean` only affects that folder. Moreover it allows running the same tests with different simulators at the same time.

nfrancque, 2022-01-13:
Thanks for the input @tasgomes.  Sounds like we all have fairly similar vunit wrappers to handle these things.

cmarqu, 2022-01-13:
If you have a handwritten setup for "entering" your project and building the environment, consider replacing it with https://modules.readthedocs.io/en/latest/ that allows to "load" and "unload" tools (by adding and removing them from `PATH`).
You could create a modulefile for simulator A that conflicts with simulator B (and the other way around), and thus have vunit automatically select the simulator that you loaded.

tasgomes, 2022-01-14:
I do have a Makefile on top of all of this, with some switches that allows me to easily start VUnit with any simulator from the console. The Makefile then manages the simulator name, simulator path, vivado libraries path, etc.

It also allows me to pass arguments to VUnit so that I can run a specific testcase in GUI mode for example.

nfrancque, 2022-01-14:
Thanks @cmarqu hadn't seen that.  We have  `startfile` that's basically just a bash script to source for the environment.

One thing, I do think it's a feature that vunit doesn't mandate a particular workflow or even really suggest one.  This has some benefits on the adoption side.  So some of these things may be better suited as remaining outside.  I'm not sure where that line falls.

LarsAsplund, 2022-01-14:
Today the simulator is selected by the [SIMULATOR_FACTORY](https://github.com/VUnit/vunit/blob/d99fe6270493532a81fe2f2cf97423b7786a03b4/vunit/sim_if/factory.py#L160). It has a [select_simulator](https://github.com/VUnit/vunit/blob/d99fe6270493532a81fe2f2cf97423b7786a03b4/vunit/sim_if/factory.py#L102) method which is called behind the scenes and returns a simulator interface class based on what is found on the `PATH` and if the `VUNIT_SIMULATOR` env variable is set.

I think the first step would be to allow the user to request a simulator interface from the factory

`my_simulator = SIMULATOR_FACTORY.get_simulator("modelsim")` 

and then pass that to the VUnit object

`vu.set_simulator(my_simulator)`

Each simulator interface has a name, for example `modelsim`, which is used to create the subdirectory under the VUnit output path. If the user is given a method to set the name to something else, for example

`my_simulator.set_name("modelsim_microsemi_2020_3")`

it would allow for several ModelSims under the same VUnit output path.

The simulator interface also has find methods that will find the simulator executable path in `PATH` or take what is provided by the `VUNIT_MODELSIM_PATH` env variable. We could add a new `set_path` metod to take care of that.

All this could also be wrapped into a single API

`vu.set_simulator("modelsim", "modelsim_microsemi_2020_3", <path/to/modelsim>)`

The second and third arguments are optional and defaults to `modelsim` and the path found by today's methods, respectively.

nfrancque, 2022-01-18:
That sounds simple enough and I think fulfills everything I am looking for.

Not sure if exposing the factory is necessary or not. But, I do have other use cases where it would be nice to have "where is x binary from y simulator?".  Right now I have an extra `_simulator_install_dir` attribute of `vu` for things like this.

This is probably best as a separate issue, but questasim also would be better off differentiated from modelsim.  It has an optimization phase between compile and sim.  I have a hack to add this as an optional stage for all simulators.