VUnit/vunit #794: VUnit reports false pass (on failing test)

ejoergensen, 2022-01-13:
Hi,

I came across a strange problem and I have been able to isolate the code such it is easily reproducible. When running in verbose mode, I clearly see that the test fails, however, the testrunner summarizes the result as pass! See the log from the test below. The attached files should reproduce the error (by changing the delay before rst is de-asserted, the tests will correctly report failure - only one out of the three tests defined is erroneously reported as pass):

```
         60000 ps - default              -    INFO - test_rst_time_out_5 is running = 5
         80000 ps - default              -    INFO - test_rst_time_out_5 is running = 5
         80000 ps - check                -   ERROR - comparing test failed - Got 0. Expected 1.
 ** Failure: Stop simulation on log level error
    Time: 80 ns  Iteration: 0  Process: /test_tb/p_main_test_runner File: C:/python/WPy64-3770/python-3.7.7.amd64/Lib/site-packages/vunit/vhdl/core/src/core_pkg.vhd
 Break in Subprogram core_failure at C:/python/WPy64-3770/python-3.7.7.amd64/Lib/site-packages/vunit/vhdl/core/src/core_pkg.vhd line 84
 Stopped at C:/python/WPy64-3770/python-3.7.7.amd64/Lib/site-packages/vunit/vhdl/core/src/core_pkg.vhd line 84

 Test Run Failed!

 Stack trace result from 'tb' command
 (....)

 Surrounding code from 'see' command
   79 :   begin
   80 :     if core_failure_is_mocked then
   81 :       set(core_failure_mock_state, core_failure_called_idx, 1);
   82 :       reallocate(to_string_ptr(get(core_failure_mock_state, core_failure_message_idx)), msg);
   83 :     else
 ->84 :       report msg severity failure;
   85 :     end if;
   86 :   end;
   87 :
   88 :   procedure check_core_failure(msg : string := "") is

pass (P=1 S=0 F=0 T=1) lib.test_tb.test_rst_time_out_5 (1.9 seconds)

==== Summary ===========================================
pass lib.test_tb.test_rst_time_out_5 (1.9 seconds)
========================================================
pass 1 of 1
========================================================
Total time was 1.9 seconds
Elapsed time was 1.9 seconds
========================================================
All passed!
```

[run.py.txt](https://github.com/VUnit/vunit/files/7864991/run.py.txt)
[test_tb.vhd.txt](https://github.com/VUnit/vunit/files/7865070/test_tb.vhd.txt)

LarsAsplund, 2022-01-13:
The problem here is the while loop

```  vhdl
while not sim_completed(2) = '1' loop
```
which causes the test case to be run twice:

``` vhdl
      elsif run("test_rst_time_out_5") then
        info("test_rst_time_out_5 is running = " &   to_string(G_RST_TIME_OUT));
```

That is why you see the info message twice

``` console
         60000 ps - default              -    INFO - test_rst_time_out_5 is running = 5
         80000 ps - default              -    INFO - test_rst_time_out_5 is running = 5
```

The failure to register the failed test is a very unfortunate side-effect of this. We need to detect this "dual test case run" and report that as an error. Meanwhile you need to change your testbench such that each test case is only run once.

ejoergensen, 2022-01-13:
Thanks a lot for the explanation - your prompt response is highly appreciated. 

It was a little odd case and it shouldn't be a big problem rewriting the testbench.

LarsAsplund, 2022-01-14:
I fixed this on `main` now in the sense that the failing test won't go undetected. This still room for improvement though by detecting the improper dual test case run.  I will open a separate enhancement issue for that.

closed 2022-01-14 by LarsAsplund