VUnit/vunit #801: regression run in own testcase directory

asicnet, 2022-02-17:
I use protected models from the suppliers.
They always write the same files. In parallel running simulation they are always overwritten. Thus a reference comparison is not possible.

I am looking for a simple solution to run the simulation not in the directory ...../vunit_out/modelsim/ with the libraries and modelsim.ini but in ...../vunit_out/modelsim/testxyz. 

or run it in the output_path

but unfortunately I can't find anything suitable

The other threads could not help

Maybe a easy solution exists

BR
Helmut

asicnet, 2022-02-18:
I have a small solution -n  vsim_simulator_mixin.py
add the test_suite_name as parameter in  _run_batch_file and define a env-variable VUNIT_UNIQUE_DIR
then i can run all tests without overwriting some files.

call is 
    return self._run_batch_file(str(batch_file_name),test_suite_name)

    def _run_batch_file(self, batch_file_name, test_suite_name, gui=False):
        """
        Run a test bench in batch by invoking a new vsim process from the command line
        """
        env = self.get_env()
        try:
            args = [
                str(Path(self._prefix) / "vsim"),
                "-gui" if gui else "-c",
                "-l",
                str(Path(batch_file_name).parent / "transcript"),
                "-do",
                f'source "{fix_path(batch_file_name)!s}"',
            ]
            sim_path=str(Path(self._sim_cfg_file_name).parent)
            if env['VUNIT_UNIQUE_DIR']:
                sim_path=sim_path+'/'+test_suite_name
                os.system("mkdir -p "+sim_path)
            proc = Process(args, cwd=sim_path)
            proc.consume_output()
        except Process.NonZeroExitCode:
            return False
        return True

Maybe it can be done with an argument  --unique_dir.

LarsAsplund, 2022-02-19:
A side effect of #785 is that we need to run every thread in a private directory. Try to check out that branch and see if it is solving your problem.

asicnet, 2022-02-19:
Ok, I have tested it. Now I get a separate directory for each process. Option -u seems to have no influence anymore. My problem is: How do I get in the post_run() and pre_config() and post_check() functions the simulator_output_path?? the test_suite_name and other information concerning the test, so I can check the correct reference data.

VG
Helmut

asicnet, 2022-02-19:
I need a possibility to copy  all generated files from the simulation_output_path to the output_path with the current test_suite_name.
BR
Helmut

LarsAsplund, 2022-02-20:
Today `simulator_output_path` is available to `pre_config` only. What you can do to share it with `post_check` is something like this.

``` python
class TestbenchInfo:
    def __init__(self, tb_name):
        self.tb_name = tb_name
        self.output_path = None
        self.simulator_output_path = None

def make_pre_config(info):
    def pre_config(output_path, simulator_output_path):
        info.output_path = output_path
        info.simulator_output_path = simulator_output_path

        # Your code

        return True

    return pre_config

def make_post_check(info):
    def post_check(output_path):
        # Your code will have all information available through the info object
        return True

    return post_check

info = TestbenchInfo("my_testbench")

tb.set_pre_config(make_pre_config(info))
tb.set_post_check(make_post_check(info))
```

You can make a `post_run` the same way the `post_check` was made

LarsAsplund, 2022-02-20:
If you're not running in parallel threads then you will still have files being overwritten. You can always copy files from `simulator_output_path` to the test unique `output_path` to avoid that.

asicnet, 2022-02-21:
Ahh  
def pre_config(output_path, simulator_output_path): 
has a second parameter.

Will this important possibility of parallization be taken over into the master branch in the future?

asicnet, 2022-02-21:
I mean simulator_output_path and own subdir for each thread

LarsAsplund, 2022-02-21:
`simulator_output_path` is already on master branch but private subdirs is an update in the near future. If you do not use parallel threads it will probably possible for you to copy data between `simulator_output_path` and `output_path` to avoid that tests overwrite each other

asicnet, 2022-02-22:
Only outputpath is in the docu in Pre and post simulation hooks[¶](https://vunit.github.io/py/ui.html#pre-and-post-simulation-hooks)

LarsAsplund, 2022-02-22:
Yeah, I see that it is missing from the documentation. I will add that. It is supported though: https://github.com/VUnit/vunit/blob/a66d757f720c40e705caabe4b360fd32e2dff072/vunit/configuration.py#L152

asicnet, 2022-02-22:
maybe the same for the undocumented "output" in line 170

closed 2022-07-10 by LarsAsplund