VUnit/vunit #803: Are the verilog config are supported ?

vperrin59, 2022-02-24:
I'm trying to use configs for some test bench:

> config tb_cfg;
>    // Define top
>    design tb_lib.tb;
>    /* rules begin */
>    default liblist lib1 lib2;
>    /* rules end */
> endconfig

I didn't see something in the doc in the function arguments that would allow that

LarsAsplund, 2022-02-25:
My Systemverilog skills are limited but it looks like the config files are added as separate command line options when running the simulator compile and simulate commands. This is something you can do with `set_compile_option`, `set_sim_option`, or `add_config`. Note that the VUnit configuration concept has no relation to these SV configs. We are in the process of creating a more tight integration between VUnit configurations and VHDL configurations and maybe that is a potential future for SV configs as well.

vperrin59, 2022-02-25:
I was referring to 3.10 of IEEE 1800-2012. I think what is missing in the current Vunit to support is the ability to define the selected config with -top switch for xcelium (That is what I'm using)

vperrin59, 2022-02-25:
For example in Verilog:

    config basic_tb_cfg;
       // Define top
       design tb_lib.tb;
       /* rules begin */
       /* rules end */
    endconfig

And then 

    xrun 
    -top cpu_sys_tb_cfg

LarsAsplund, 2022-02-27:
If it comes down to using the config as the top-level for the simulation it is the same type of problem that we're about to solve for VHDL configurations. To support that for SV we need to identify SV configs and figure out if they are associated with a VUnit testbench. If so, we create a VUnit configuration for that testbench which holds the information needed to change the behavior of VUnit from simulating the testbench to simulating the configuration. In your example we would create a VUnit configuration named `basic_tb_cfg` for the testbench `tb`.