VUnit/vunit #807: Generate vhdl database file for LSP's

delafthi, 2022-03-18:
What do you guys think about a feature to generate the database file for LSP? This functionality would be similar to the `CMAKE_EXPORT_COMPILE_COMMANDS` option in CMake, enabling the output of the `compile_commands.json` file for clangd.

I imagine a method called on the vunit object instance, which outputs all files currently belonging to the project. Something like this:

```python
vu = VUnit.from_argv()
lib = vu.add_library("lib")
lib.add_source_files("*.vhd")

# Create the db file here
vu.create_db(type="vhdl_ls")

vu.main()
```

The type argument should configure the output format to the specific lsp, for exmple [vhdl_ls](https://github.com/VHDL-LS/rust_hdl) or [ghdl-ls](https://github.com/ghdl/ghdl-language-server).

Currently, I use [vhdl_ls](https://github.com/VHDL-LS/rust_hdl) as an LSP and use the following workaround:

```python
def _gen_vhdl_ls(vu, output_file):
    """
    Generate the toml file required by rust_hdl (vhdl_ls).

    Args:
        vu: A VUnit object file.
        output_file: A string containing the path to the output file.
    """
    proj = vu._project
    libs = proj.get_libraries()
    vhdl_ls = {"libraries": {}}
    for lib in libs:
        vhdl_ls["libraries"].update(
            {lib.name: {"files": [file for file in lib._source_files]}}
        )
    with open(output_file, "w") as f:
        toml.dump(vhdl_ls, f)
```

LarsAsplund, 2022-04-09:
The general approach to VUnit has been to provide interfaces allowing tool integrations but not provide the actual integrations. What's missing now is a method `get_libraries` for the VUnit object. Then `get_source_files` can be used on all library objects.

delafthi, 2022-04-09:
So your suggested approach would be to implement `get_libraries` method for the VUnit class and leave the generation of the database file up to the user.

A possible implementation in `vunit/ui/__init__.py` would be:

```python
def get_libraries(self):
    return [self.library(lib.name) for lib in self._project.get_libraries()]
```

which returns a list of `vunit.ui.library.Library` objects, on which `get_source_files` could be called. But maybe there is a better way to convert the `vunit.library.Library` objects to `vunit.ui.library.Library` objects. Additionally, the output of `self._project.get_libraries()` returns an odict. Would it be better to keep the return value of the `get_libraries` method also an odict instead of a regular list?

LarsAsplund, 2022-04-16:
Yes, something along these lines (I have to recall why it is an odict). I do not suggest that every user should do this from scratch. It would be great if there is some open source solution for it. I'm just saying that such a solution should be separate from the VUnit repository. Such a repository could even reside under the VUnit organization in GitHub but it would have it's own release cycle and ideally maintainer.

LarsAsplund, 2022-04-17:
I started to look at this and I think we should return a new class `LibraryList` which inherits from Python `list`. In addition `LibraryList` will provide applicable methods from `Library`.

LarsAsplund, 2022-04-19:
There is now a new [get_libraries](https://vunit.github.io/py/vunit.html#vunit.ui.VUnit.get_libraries) method that you can use. Do you think this is sufficient to close this issue?

delafthi, 2022-04-19:
Thanks for the quick implementation. I did not yet look into it, because I was offline during easter. However, I'll test the implementation on my current project this week.

LarsAsplund, 2022-04-19:
Please do and give your feedback. If it works as you expect I will close this issue.

delafthi, 2022-04-20:
For my purpose, the new `LibraryList` class worked as expected. However, I did not test all other class methods, but I had a quick look into your changes, and to me, they seem fine.

Here is the function I use with #815 merged to generate my database.

```python
def _gen_vhdl_db(vu, output_file):
    """
    Generate the toml file required by rust_hdl (vhdl_ls).

    Args:
        vu: A VUnit object file.
        output_file: A string containing the path to the output file.
    """
    libs = vu.get_libraries()
    vhdl_ls = {"libraries": {}}
    for lib in libs:
        vhdl_ls["libraries"].update(
            {
                lib.name: {
                    "files": [file.name for file in lib.get_source_files()]
                }
            }
        )
    with open(output_file, "w") as f:
        toml.dump(vhdl_ls, f)
```

Maybe I will create a repo containing some code to generate the database files. However, currently, I'm a little bit short on time. So this must wait. I will reach out to you If I have something.

LarsAsplund, 2022-04-20:
That's ok, I have tests for everything but I just wanted to make sure that your use case is covered.

closed 2022-04-20 by LarsAsplund