VUnit/vunit #809: Queue_vec_t behaves like a single queue instead of a vector of queues.

DominiqueMeus, 2022-04-01:
Hi,

When making a vector of queue's using the queue_vec_t type from the queue_pkg.vhd I noticed that the underlying p_meta value is the same for all indices in the array. In the following example I expected to create 2 separate queues and the ability to push and pop from them separately. Instead it seems like I am pushing to the same queue no matter which index I use.
```
library vunit_lib;
context vunit_lib.vunit_context;
context vunit_lib.vc_context;

entity queue_vec_t_broken_example is
    generic(runner_cfg : string);
end entity queue_vec_t_broken_example;

architecture RTL of queue_vec_t_broken_example is
begin
    main : process is
        variable queue_vec : queue_vec_t(1 to 2) := (1 to 2 => new_queue);
    begin
        test_runner_setup(runner, runner_cfg);
        while test_suite loop
            if run("test_queue_vec_t") then
                push_integer(queue_vec((1)), 1);
                push_integer(queue_vec((2)), 2);
                report "p_meta of 1 " & to_string(queue_vec(1).p_meta);
                report "data of 1 " & to_string(queue_vec(1).data);
                report "p_meta of 2 " & to_string(queue_vec(2).p_meta);
                report "data of 2 " & to_string(queue_vec(2).data);
                check_equal(pop_integer(queue_vec(2)),2,"queue returned incorrect value", warning); -- Returns 1
                check_equal(pop_integer(queue_vec(1)),1,"queue returned incorrect value", error); -- Returns 2
            end if;
        end loop;
        test_runner_cleanup(runner);
    end process main;
end architecture RTL;
```
0 ps - check                        - WARNING - queue returned incorrect value - Got 1. Expected 2.
0 ps - check                        -   ERROR - queue returned incorrect value - Got 2. Expected 1.

If you use the syntax (new_queue, new_queue) instead to initialize queue_vec then everything works as expected. However this is rather cumbersome when trying to initialize many queues.

I am not well-versed in vhdl library writing but it seems like when you use the syntax others => or x to y => you get copies of identical objects while explicit instantiation provides unique objects.

Regards,
Dominique

LarsAsplund, 2022-04-09:
You are correct, this is how VHDL works. `new_queue' is only called once and is assigned both elements. A helper function `new_queue_vec` would be the way around this.

DominiqueMeus, 2022-04-09:
```
-- Queue_pkg.vhd
impure function new_queue_vec (
  size : natural
  ) return queue_vec_t;

-- Queue_pkg-body.vhd
impure function new_queue_vec (
  size : natural
  ) return queue_vec_t is
    variable ret_val : queue_vec_t(size - 1 downto 0);
  begin
    for i in 0 to size - 1 loop
      ret_val(i) := new_queue;
    end loop;
	return ret_val;
  end;
```

DominiqueMeus, 2022-04-09:
Don

LarsAsplund, 2022-04-16:
Yes, that is what I had in mind. In this case we could also consider making a `queue_pkg-2019p.vhd` file just like there is a `queue_pkg-2008p.vhd` file. The new file would include a function

``` vhdl
function new_queue_vec return result of queue_vec_t is
  variable ret_val : queue_vec_t(result 'length - 1 downto 0);
begin
  for i in 0 to size - 1 loop
    ret_val(i) := new_queue;
  end loop;
  
  return ret_val;
end;