VUnit/vunit #810: Error after updating ModelSim `ieee.std_logic_1164, ieee.numeric_std have changed.`

oscargus, 2022-04-06:
Minor issue that is easily solved by just removing the `vunit_out` directory, but thought I'd make a note of it in case it is also easy to solve in vunit (and it is not that often that one can come across it).

I guess the actual problem is `ieee.std_logic_1164, ieee.numeric_std have changed.`, but this happened after I had updated ModelSim. I assume it may happen in other situations as well.

(Obfuscated actual top level path and testbench name.)
```
=== Command output: ===
** Error: (vcom-13) Recompile vunit_lib.string_ops because ieee.std_logic_1164, ieee.numeric_std have changed.
** Error (suppressible): XXX_testbench.vhdl(4): (vcom-1195) Cannot find expanded name "vunit_lib.string_ops".
** Error: XXX_testbench.vhdl(4): Unknown expanded name.
init_auxdbinfo(): INTERNAL ERROR: Cannot open page file "XXX_/vunit_out/modelsim/libraries/vunit_lib/_lib1_0.qpg" in read mode
doVOpenFlatFile(): INTERNAL ERROR: Cannot obtain auxid to openXXX_/vunit_out/modelsim/libraries/vunit_lib/vunit_context/_primary.dat
** Error: XXX_testbench.vhdl(5): (vcom-1915) Cannot load design unit "vunit_lib.vunit_context".
** Error: (vcom-7) Failed to open library file "XXX_/vunit_out/modelsim/libraries/vunit_lib/vunit_context/_primary.dat" in read mode.
No such file or directory. (errno = ENOENT)
** Note: XXX_testbench.vhdl(5): VHDL Compiler exiting
```

LarsAsplund, 2022-04-06:
What you can do is to run your run script with the `--clean` option. That we recompile everything from scratch. Sometimes necessary in situations like yours.

oscargus, 2022-04-09:
Thanks for the answer! The point was that maybe one could infer the error message and then recompile automatically. Or possibly output an error message suggesting `--clean`.

Another, slightly related, issue is that when running `--clean` I got an error that there were some `.nfs`-files that could not be removed. This is clearly a problem on our side, but again, the error message could potentially be a bit more helpful. (Should have stored the log for this as I guess a simple try-except could catch it and output a more easy to understand warning. Maybe there will be a PR.)

But it all comes down to effort in detecting all possible corner-cases and the probability that one guesses the correct one.

LarsAsplund, 2022-04-16:
The problem here is that VUnit doesn't know that a recompile is needed because of an updated simulator version. It is ModelSim that outputs these error messages. Internally VUnit see if files have changed or if the files they depend of have changed. This is what causes a recompile. I guess we could include monitoring of the simulator version as well but I feel that might be a bit out of scope for VUnit since it cannot provide a fully waterproof solution. If you have a pre-compiled vendor library added to VUnit with `add_external_library` we do not have the information to determine if it has been compiled with a version compatible with the current simulator or not.

LarsAsplund, 2022-07-16:
I've added a bit more information on the help for the `--clean` option. I think it is the best I can do in this case.

PR #855 (2022-07-16): Improved --clean help with example.example
Closes #810.
commit 57639004 (2022-07-16): Improved --clean help with example.example

Closes #810.
closed 2022-07-16 by LarsAsplund