VUnit/vunit #812: Add peek functionality to the `queue_pkg`

dalex78, 2022-04-17:
__Objective:__
Add a `peek` functionality to the `queue_pkg`: returns the element at the front the queue. It does not deletes the element in the queue (contrary to the already existing `pop` command).

__Rational:__
I intend to use this feature to improve the AXI(L) verification components (I understand from issue #726 that the development of VC is in stand-by, any news regarding that ?).

__What has been done:__
 - The `peek` functionality has been implemented here: [peek_functionality](https://github.com/dalex78/vunit/tree/peek_functionality)
 - I have launched the following tests on Modelsim (Interl FPGA Starter Edition 2020.1): all `run.py` inside the `vunit/vhdl` architecture:
 - + Some of the tests of the `verification_components` have failed but are not related to this work (see issue #642).

__Remaining work__: (if this feature is wanted/accepted by the community)
 - Launch all tests with tox on modelsim (currently doing that),
 - Test with other simulators,
 - Update the documentation

__Additional remarks:__
 - I think the `codec_builder_pkg` and `queue_pkg` files deserve a little bit of cleanup: more documentation, removing hard-coded values, etc (without breaking compatibility). I can work on that before doing a pull request, or it could be done at a later time.

This is my first (code) contribution, don't hesitate to give feedback.

LarsAsplund, 2022-04-17:
Agree that a peek is useful. I thought about if when fixing #642. Note that we have something similar in `com` (https://github.com/VUnit/vunit/blob/4a0e25830ad28f9ed2009c3e2da426cae3be2e40/vunit/vhdl/com/test/tb_com.vhd#L937)

You're right, the facelift of all VCs has been on hold for a while. The last time I worked on it I presented the standard rules in the Gitter channel and the feedback was such that we can proceed. There is one thing that I will revisit and maybe do a bit differently. This is also a backward compatibility breaking update so it we need to be a bit more careful when deciding on when to introduce it.

I suggest you go ahead and make a PR of peek for queues.

dalex78, 2022-04-17:
Corresponding PR: #814