VUnit/vunit #831: Feature Request: Allow to add *.gtkw files to a testbench for GTKWave

sudden6, 2022-05-18:
I want to store "GTKWave" save files along with my testbenches, so that I don't have to reconfigure the view all the time.

This is possible by executing `run.py --gtkwave-args "-a <path to gtkw file>.gtkw" <testbench name>` via command line, but I didn't find a way to pass a per testbench `*.gtkw` file via `run.py` so that someone else can easily use my preconfigured waveform views.

LarsAsplund, 2022-05-29:
@sudden6 Will `ghdl.gtkwave_script.gui` help? See https://vunit.github.io/py/opts.html#simulation-options.

sudden6, 2022-05-29:
In principle it can do the same thing, the advantage of `*.gtkw` files is, that GTKWave can write its current GUI setup itself, so updating them is very easy and user friendly.

I just realized another way to achieve this and maybe even more is to make the `--gtkwave-args´ and `--gtkwave-fmt` run time arguments configurable per testbench. It seems other simulator have this but not GTKWave.