VUnit/vunit #833: "Package previously defined" warnings for OSVVM packages

bradleyharden, 2022-05-27:
We're seeing the following warnings on every invocation of VUnit.

```
WARNING - /usr/local/lib/python3.6/dist-packages/vunit/vhdl/osvvm/ScoreboardPkg_int_c.vhd: package 'scoreboardpkg_int' previously defined in /usr/local/lib/python3.6/dist-packages/vunit/vhdl/osvv
m/ScoreboardPkg_int.vhd                                                                                                                                                                            
WARNING - /usr/local/lib/python3.6/dist-packages/vunit/vhdl/osvvm/ScoreboardPkg_slv_c.vhd: package 'scoreboardpkg_slv' previously defined in /usr/local/lib/python3.6/dist-packages/vunit/vhdl/osvv
m/ScoreboardPkg_slv.vhd
```

I see some other issues related to these packages, but none with the same warning. Any idea what the cause might be?

LarsAsplund, 2022-05-28:
@bradleyharden  This looks like the problem we fixed in #769. This is not part of a release yet, you need to use latest and greatest or wait for the v4.7.0 release. Are you using v4.6.0?

bradleyharden, 2022-05-28:
Yes, I'm on 4.6. I don't know why I didn't think to try `master` before creating an issue. I'll do that.

bradleyharden, 2022-06-02:
I can confirm this is fixed on `master`. Sorry for the spam.

closed 2022-06-02 by bradleyharden