VUnit/vunit #834: Add 'questa' to list of supported simulators (VUNIT_SIMULATOR)

tasgomes, 2022-05-30:
Currently, if `VUNIT_SIMULATOR` is set to 'questa' the following error occurs:

> RuntimeError: Simulator from VUNIT_SIMULATOR environment variable 'questa' is not supported. Supported simulators are dict_keys(['modelsim', 'rivierapro', 'activehdl', 'ghdl', 'incisive'])

However, both Questa and Modelsim call `vsim` to run the simulations. In fact, if I set `VUNIT_SIMULATOR='modelsim'` and `VUNIT_MODELSIM_PATH=<path to questa install directory>`, I can execute VUnit testbenches with Questa without any problem.

Would it be possible to add the name 'questa' to the current list of supported simulators? 

or is there any limitation?

umarcor, 2022-05-30:
Would the enhancement include supporting `VUNIT_QUESTASIM_PATH` as an alias of `VUNIT_MODELSIM_PATH`?

Context: https://github.com/VUnit/vunit/blob/fbb46c60b3b6b150bfaeb6e4fb30afa8ccdaf5a1/vunit/sim_if/factory.py#L102-L127

tasgomes, 2022-05-30:
@umarcor There are also some other considerations, example:

`lib.set_sim_option("modelsim.vsim_flags", (...))`

tasgomes, 2022-05-31:
@std-max as far as my experience goes. with Questa-Intel FPGA Edition, the file is still called `modelsim.ini`.

If I compare the `vunit_out` folders between Questa and Modelsim:

- Questa: `.\vunit_out\modelsim\modelsim.ini`
- Modelsim: `.\vunit_out\questa\modelsim.ini`

tasgomes, 2022-06-01:
Currently all my (minor) issues can be solved with workarounds. But I am wondering if we are delaying something that might need to be done in the future.

At the moment, I have a Makefile that allows me to change the simulator (modelsim, questa and ghdl) using a switch. This switch is also used for Vivado to compile the Xilinx libraries where Vivado distinguishes between Modelsim (`-simulator modelsim`) and Questa (`-simualtor questa`). Because I compile the libraries to specific folders and use different `vunit_out` folders depending on the simulator used, I can run the same regression test in any of the simulators at the same time.

At the moment, one of the workarounds in `run.py` is to overwrite the name of the simulator when `questa` is select as follow:

```python
if args.simulator == 'questa':
    environ["VUNIT_SIMULATOR"] = 'modelsim'
else:
    environ["VUNIT_SIMULATOR"] = args.simulator
```

One main difference between Questa and Modelsim, is that Questa has optimizations turned on by default. If you try to disable them, you even get a (suppressible) compilation error that your simulation will run very slowly, etc.. The point is because optimizations are on, if you debug your DUT in GUI mode many internal signals are optimized out and not visible so you need to add some flags to vsim in order to preserve visibility for debugging.

Also, comparing the help menu of `vsim` and `vcom` shows that there many new flags in Questa that do not exist in Modelsim so at some point it will make sense to have `modelsim.vsim_flags` and `questa.vsim_flags` like we have for `ghdl.elab_flags` so that new Questa flags are not applied when using Modelsim simulator. Again this can still be solved with the workaround shown above.

LarsAsplund, 2022-07-10:
@tasgomes Can you share some of the flags that are different? Just to get a feel for how "useful" they are. 

Some weeks ago a new colleague tried to install free ModelSim from Intel to use that with VUnit. He found out that the free simulator in Quartus is now a variant of Questa which he didn't manage to install. Not sure if it wasn't free after all. Anyway, it was solved by using an older version of Quartus to get access to the free ModelSim.

If there actually is a free Questa from Intel it means that it will be easier for us to verify Questa and also support it as a simulator different from ModelSim. Anyone having experience with this new Questa version?

tasgomes, 2022-07-11:
@LarsAsplund Although there are now new flags in Questa, currently there is none that I used that only exists in one of them. But since I started using Questa, I had to add this one to preserve signal visibility when doing debug activities in GUI mode:

```
lib.set_sim_option("modelsim.vsim_flags", ["-voptargs=+acc"])
```

I use Questa Intel FPGA Edition. I believe when you are installing it, you can select the free "Starter Edition" option.

closed 2022-07-11 by LarsAsplund

tasgomes, 2025-05-21:
@thrakkor I have the same issue. I solved it as described above https://github.com/VUnit/vunit/issues/834#issuecomment-1143183668. I created a custom argument for the run.py that sets the simulator name being used (see above `args.simulator`). Depending on this argument, I conditionally select the source for the vendor libs.

I ran into the same case where multiple projects might use different simulators, or even different projects can use different versions of the same simulator, hence I need to handle this per project.

My makefile will compile the vendor libs no matter the simulator version, and link it correctly with the run.py. I also have another custom argument to specify the location of the compiled vendor libs.

PR #850 (2022-07-11): Clarify that VUNIT_SIMULATOR is set to modelsim when using Questa.
Closes #834
commit 55b15a68 (2022-07-11): Clarify that VUNIT_SIMULATOR is set to modelsim when using Questa.

Closes #834