VUnit/vunit #842: Update OSVVM

GlenNicholls, 2022-06-08:
I'm getting an error with ActiveHDL 11.1:

```
=== Command output: ===
COMP96 WARNING COMP96_0048: "This function may complete without return statement." "C:/Users/Glen.Nicholls/AppData/Local/Programs/Python/Python37/Lib/site-packages/vunit/vhdl/osvvm/ScoreboardPkg_slv_c.vhd" 1795 5
COMP96 WARNING COMP96_0048: "This function may complete without return statement." "C:/Users/Glen.Nicholls/AppData/Local/Programs/Python/Python37/Lib/site-packages/vunit/vhdl/osvvm/ScoreboardPkg_slv_c.vhd" 2225 5
COMP96 ERROR COMP96_0106: "Returned value type does not match the type declared in function declaration." "C:/Users/Glen.Nicholls/AppData/Local/Programs/Python/Python37/Lib/site-packages/vunit/vhdl/osvvm/ScoreboardPkg_slv_c.vhd" 2619 12
COMP96 ERROR COMP96_0106: "Returned value type does not match the type declared in function declaration." "C:/Users/Glen.Nicholls/AppData/Local/Programs/Python/Python37/Lib/site-packages/vunit/vhdl/osvvm/ScoreboardPkg_slv_c.vhd" 2629 12
COMP96 Compile failure 2 Errors 2 Warnings  Analysis time :  0.1 [s]
```

After updating lines 2612-2630 to the following (OSVVM latest), it fixes the issue:

``` vhdl
impure function Peek (
    constant ID     : in  ScoreboardIDType ;
    constant Tag    : in  string
  ) return ExpectedType is
  begin
--      return ScoreboardStore.Peek(Tag) ;
    log("Issues compiling return later");
    --return ScoreboardStore.Peek(ID.ID) ;
    return ScoreboardStore.Peek(Index => ID.ID, Tag => Tag) ;
  end function Peek ;

  -- Simple Scoreboard
  impure function Peek (
    constant ID     : in  ScoreboardIDType
  ) return ExpectedType is
  begin
    --return ScoreboardStore.Peek(ID.ID) ;
    return ScoreboardStore.Peek(Index => ID.ID) ;
  end function Peek ;
```

umarcor, 2022-06-23:
@GlenNicholls I guess you are using the latest release of VUnit but not the `master`? See #769 and #833.

GlenNicholls, 2022-06-23:
Correct, for the issue I describe, I installed VUnit with pip.

yourfate, 2022-07-13:
I am seeing the same error with aldec riviera + vivado:

```
2022-07-13 13:41:04,671 - CTR - ERROR - Compiling into osvvm:     .venv\Lib\site-packages\vunit\vhdl\osvvm\ScoreboardPkg_slv_c.vhd                               failed
2022-07-13 13:41:04,671 - CTR - ERROR - === Command used: ===
2022-07-13 13:41:04,671 - CTR - ERROR - C:\Aldec\Riviera-PRO-2020.04-x64\bin\vcom -quiet -j C:\workspace\dsp-emulation\build\vunit_out\rivierapro -dbg -2008 -work osvvm C:\workspace\projectname\.venv\Lib\site-packages\vunit\vhdl\osvvm\ScoreboardPkg_slv_c.vhd
2022-07-13 13:41:04,671 - CTR - ERROR -
2022-07-13 13:41:04,671 - CTR - ERROR - === Command output: ===
2022-07-13 13:41:04,671 - CTR - ERROR - COMP96 WARNING COMP96_0048: "This function may complete without return statement." "C:\workspace\projectname\.venv\Lib\site-packages\vunit\vhdl\osvvm\ScoreboardPkg_slv_c.vhd" 1795 5
2022-07-13 13:41:04,671 - CTR - ERROR -
2022-07-13 13:41:04,671 - CTR - ERROR - COMP96 WARNING COMP96_0048: "This function may complete without return statement." "C:\workspace\projectname\.venv\Lib\site-packages\vunit\vhdl\osvvm\ScoreboardPkg_slv_c.vhd" 2225 5
2022-07-13 13:41:04,671 - CTR - ERROR -
2022-07-13 13:41:04,671 - CTR - ERROR - COMP96 ERROR COMP96_0106: "Returned value type does not match the type declared in function declaration." "C:\workspace\projectname\.venv\Lib\site-packages\vunit\vhdl\osvvm\ScoreboardPkg_slv_c.vhd" 2619 12
2022-07-13 13:41:04,671 - CTR - ERROR -
2022-07-13 13:41:04,671 - CTR - ERROR - COMP96 ERROR COMP96_0106: "Returned value type does not match the type declared in function declaration." "C:\workspace\dsp-emulation\.venv\Lib\site-packages\vunit\vhdl\osvvm\ScoreboardPkg_slv_c.vhd" 2627 12
2022-07-13 13:41:04,671 - CTR - ERROR -
2022-07-13 13:41:04,671 - CTR - ERROR - COMP96 Compile failure 2 Errors 2 Warnings  Analysis time :  31.0 [ms]
2022-07-13 13:41:04,671 - CTR - ERROR -
2022-07-13 13:41:04,671 - CTR - ERROR -
2022-07-13 13:41:04,671 - CTR - ERROR - Compile failed
```

I have vunit_hdl 4.6.0 from pip.

LarsAsplund, 2026-11-02:
This is an old issue that is no longer relevant. Since OSVVM has been moved to the separate [vunit-osvvm package](https://github.com/VUnit/vunit-osvvm), any future issues should be reported in that repository.

closed 2026-11-02 by LarsAsplund