VUnit/vunit #846: Where to find documentation and examples for the AXI4 Lite bus master verification component?

creiter64, 2022-06-28:
The documentation in https://vunit.github.io/verification_components/user_guide.html mostly shows what's available, but not really how to use theses verification components.

I have also found this example: https://github.com/VUnit/vunit/tree/master/examples/vhdl/axi_dma/src/test but it is only minimally documented.

gyuunyuu1989, 2022-07-03:
You could just read the source code of the BFM besides using the example.

creiter64, 2022-07-04:
@gyuunyuu1989 I'm aware that this is possible thanks to VUnit being open source. It would still be much easier and much more accessible when there are simple examples.

rftafas, 2022-07-04:
You can check mine: https://github.com/rftafas/stdcores/blob/master/aximm_intercon/aximm_intercon_tb.vhd