VUnit/vunit #848: VCI using protected types (VHDL2008)

rftafas, 2022-07-05:
I've been using protected types to create a few VCIs (currently, I2C for a 'ram' component and I2S). It was meant to be used as this, but could surely be packed in an entity for more usual experience, I mean, create an entity and add the protected variables within it.

BUT... it forces VHDL2008. I'm not sure this is desired.

Any thoughts?

A note before I get asked for WHY, I did it because I wanted to try Protected Types, i.e., learning. The example (which is an eternal WIP) is here: https://github.com/rftafas/stdcores/blob/master/i2cs_axim/i2cm_vci_pkg.vhd

umarcor, 2022-07-05:
VUnit VCIs do use protected types already. The default in VUnit is to use VHDL 2008. VHDL 1993 is supported, but most of the advanced features do require VHDL 2008. See https://github.com/VUnit/vunit/blob/a66d757f720c40e705caabe4b360fd32e2dff072/vunit/vhdl/com/src/com_messenger.vhd or https://github.com/VUnit/vunit/blob/a66d757f720c40e705caabe4b360fd32e2dff072/vunit/vhdl/verification_components/src/axi_slave_private_pkg.vhd.

rftafas, 2022-07-05:
I think I missed that one. Thanks for the links and explanation. 

Is it also available as the public 'api'? Because actually I should have been more specific to my question: instead of the VCI being an entity, it could be a protected type. Would that be ok? It stretches the definition of 'component'...

Not to argue approaches though, as potentially both will have its benefits and drawbacks.