VUnit/vunit #856: AXI4 Lite master hangs on wait_until_idle() if bus is really idle

abyszuk, 2022-07-18:
Hello,

It seems that there's a problem with the way AXI4 Lite master detects bus idle condition.
To trigger this bug it's enough to have this code in your testbench:
```vhdl
read_bus(net, bus_handle, bus_addr, bus_data);
wait_until_idle(net, bus_handle);
```
It seems to me that the problem is with [this line](https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/src/axi_lite_master.vhd#L64) of code:
`wait until ((bvalid and bready) = '1' or (rvalid and rready) = '1') and is_empty(message_queue) and rising_edge(aclk);`
This code checks only for transition from busy to idle.
If the `wait_until_idle()` is called when the bus is idle then the first part of `if` statement is impossible to meet.

Considering how the `bus_process` works, maybe it could be replaced with:
`wait until bready = '0' and rready = '0' and is_empty(reply_queue) and is_empty(message_queue) and rising_edge(aclk);`

std-max, 2022-11-05:
I had a similar problem at work with the same verification component.
I reworked it a bit but forgot to push the modifications.
I wanted to push the changes but it seems that PR dealing with VC bugfix do not really get merged :/
Hopefully in the next months there will be a repo just for VUnit VC

LarsAsplund, 2022-11-07:
@abyszuk I merged #797 which should fix this problem.

closed 2022-11-07 by LarsAsplund