VUnit/vunit #863: --xunit-xml-format does not change output

goddisignz, 2022-10-24:
I am currently trying to combine vunit with bamboo using the xunit output of VUnit and the xUnit.net Parser.
Intrerestingly, the parser always reports a "skipping non-xunit xml file [...]vunit/vunit.xml".

I tried setting the --xunit-xml-format argument to bamboo but the output file does not seem to change.
Looks like the option is ignored. When I set it to something wrong e.g. "bambooze" VUnit reports:
`run.py: error: argument --xunit-xml-format: invalid choice: 'bambooze' (choose from 'jenkins', 'bamboo')`

The xunit.xml file looks like this:
```
<testsuite name="testsuite" errors="0" failures="0" skipped="0" tests="1" hostname="HORTS"><testcase classname="xil_defaultlib.testbench_tb" name="first_test" time="1.7"><system-out>
# Lots of modelsim text
</system-out></testcase></testsuite>
```

LarsAsplund, 2022-10-25:
@goddisignz Hi, VUnit only handles failing test cases differently. Passing test cases will look the same in both formats. What is xUnit.net parser expecting?

[image]

goddisignz, 2022-10-25:
Actually, I have no clue what xUnit.net parser expects. Even if I have no failing tests, the xml-file is not accepted as xunit. As it is a bamboo plugin I cannot get any further output.
I will try to find some kind of standalone executable to obtain more info.

For completeness, I am currently using VUnit 4.6.0 installed via pip.

goddisignz, 2022-10-25:
I am getting the feeling that xUnit.net is somehow not XUnit. But it seems to be the only plugin that atlassian offers for bamboo that contains the word "xUnit". Maybe I have to find another way to integrate VUnit with our Bamboo installation. Any recommendations?

Furthermore, I will generate some failing tests and check again if the bamboo-argument makes a difference.

goddisignz, 2022-10-25:
You are right, with failing tests, my xunit output also changes. Thus, my issue is void.

Concerning my setup, I tried the JUnit test parser as I read that xunit originated from junit. Surprise, it works.
Now I feel a little bit dumb but also a bit wiser.

I hope I didn't waste too much of your time.

closed 2022-10-25 by goddisignz

LarsAsplund, 2022-10-25:
No worries, the JUnit/xUnit format is a bit of a mess since it's not standardized (at least it didn't used to be). That plugin you tried might attempt to implement xUnit but with yet another style.