VUnit/vunit #864: The max. length of runner_cfg is defined by runner_cfg_default length (Xcelium)

VinieBerry, 2022-10-31:
In many VHDL examples such as _check_ the `runner_cfg` generic has the `runner_cfg_default` default value. This generic value includes the simulation output path based on the run.py file location (vunit_cli.py).

However, during simulation I get the following error (Xcelium 21.09):
```
cannot open '/work/sesavunit_results' for writing (Permission denied)

          File: /work/sesa595203/vunit/vunit/vhdl/core/src/core_pkg.vhd, line = 32, pos = 48
         Scope: @vunit_lib.core_pkg:setup
          Time: 0 FS + 0
```
Yet, the arguments provided to the simulators are okay:
```
-generic "tb_example.runner_cfg => \"active python runner : true,enabled_test_cases : ,output path : /work/sesa595203/vunit/examples/vhdl/check/vunit_out/test_output/lib.tb_example.all_7b5933c73ddb812488c059080644e9fd58c418d9/,tb path : /work/sesa595203/vunit/examples/vhdl/check/,use_color : true\""
```
Actually, the issue seems that the `runner_cfg` string length is constrained by the `runner_cfg_default` string length, itself defined by its content set in run_types.vhd file (74 chars) :
```
constant runner_cfg_default : string := "enabled_test_cases : __all__, output path : , active python runner : false";
```
Therefore, the `runner_cfg` value is cropped:
```
active python runner : true,enabled_test_cases : ,output path : /work/sesa
```
How should we deal with this issue ?

std-max, 2022-11-04:
With which VHDL standard do you compile ?

LarsAsplund, 2022-11-04:
The default value supports the standalone mode to run a testbench without a Python runner. That's very much a corner case today so we could really remove the default from the examples.

However, there are coming features (#772) which adds other defaults so in general this needs fixing. We could add an end marker like `NUL` to string and then pad with additional characters to extend the length. The problem is to figure out how long is long enough. I recall that there was a simulator that had a limit in the past. Not sure what the status is today.