VUnit/vunit #865: axi_stream_checker rule 10

Nik-Sch, 2022-11-01:
The rule 10 of the `axi_stream_checker` makes sure that the user data is not `X` when the `aresetn` is high. This is also in compliance with the [AMBA AXI4-Stream Protocol Assertions User Guide](https://developer.arm.com/documentation/dui0534/b/Protocol-Assertions-Descriptions/AXI4-Stream-protocol-assertion-descriptions).
However, this doesn't seem to make sense to me and the [Xilinx AXIS protocol checker](https://docs.xilinx.com/v/u/en-US/pg145-axis-protocol-checker) seems to have omitted the test. In other implementations I have [found](https://github.com/jkopanski/802.15.4/blob/master/doc/amba/sva/Axi4StreamPC.sv) the assertion is changed to only be done when `tvalid` is high.
I would suggest to either remove the assertion all together or to alter the assertion to only be performed when `tvalid` is high.

What are your thoughts on this check?

LarsAsplund, 2022-11-04:
Some users want to make sure that there are no `X` produced by the design at any point so changing the behavior to ignore this would prevent that and break backward compatibility of the protocol checker. What you can do is to disable the check, or more precisely the logger reporting errors:

``` vhdl
disable(get_logger("axi_stream_pkg:rule 10"), error);
```

To avoid using this by mistake there is another safety mechanism you have to override:

``` vhdl
test_runner_cleanup(runner, allow_disabled_errors => true);
```

Nik-Sch, 2022-11-07:
Ok, that sounds plausible and the workaround works for me.

closed 2022-11-07 by Nik-Sch