VUnit/vunit #874: Question: How to conviniently check that signal is at expected level on all clock edges?

creiter64, 2022-11-14:
Hi all,

This question might be trivial to you, but I can't wrap my head around how this is supposed to work.

I want to check that a signal is at `'0'` on every rising clock edge of the testbench. For this I found the concurrent function invocation of `check` from this [example](https://github.com/VUnit/vunit/blob/master/examples/vhdl/check/tb_example.vhd#L345).

Now I want it to always be enabled, so I call `status_check: check(clk, '1', status_ok, "Concurrent status check failed.");` but then ModelSim gives me an error that `en` must be a signal name. Do I always have to create a dummy signal which is `'1'`? Am I using the correct function for this job? Can `check(...)` be expanded to allow this case?

Thanks in advance!

LarsAsplund, 2022-11-14:
@creiter64 There is a predefined signal [`check_enabled`](https://github.com/VUnit/vunit/blob/3292c28d92798981ba5364c0f7c9020b13db7ee6/vunit/vhdl/check/src/check_api.vhd#L23) that you can use. I see that it is not mentioned in the documentation. I will update that.

closed 2022-11-14 by LarsAsplund

creiter64, 2022-11-14:
Thank you for the quick response!

commit ce8db50e (2022-11-14): Fixes #874