VUnit/vunit #876: Modelsim PE 10.6d vs Questa

apopa, 2022-11-23:
Hi,

I am having a test working in Modelsim but fails in Questa. My supposition is that the transaction is fetched from the queue by bus_master.
Did somebody encounter this kind of issue? Do you know how to debug it? The VHDL version is 2002. 

Thanks,
Adrian

LarsAsplund, 2022-11-23:
We had a number of issues being discussed on Gitter (https://gitter.im/VUnit/vunit) lately related to problems with Questa. Since no maintainer has access to Questa it can be a bit hard to debug from our side. I'm working on this access issue though. Could you start by sharing your Questa version and a minimal code example that fails and a log på the error messages you get, if any. That way I may be able to give you some ideas on what to debug. Even if I can reproduce maybe others can.

apopa, 2022-11-23:
Hi, I will ask if I can share any code. Anyways I do not think I can provide you the full BFM. To refine the question, should I go to VHDL 2008 instead of 2002?
The logs will not help. The error is generated by unexpected_msg_type.

Thanks,
Adrian

LarsAsplund, 2022-11-23:
VHDL 2002 is a bit strange. The `com` package used by `bus_master` is only supported for VHDL 2008 and later. How do you compile the code? If you use VUnit for that you should have a call to the `add_com` method in your run script. That method will check that you have the correct standard and fail if that is not the case.

LarsAsplund, 2022-12-09:
@apopa Any news on this? I got access to a Questa license and ran our test suite for the bus master. It works and I suggest that you do the same to see if we can zoom in on the problem without any of your proprietary code. Go to \<vunit install directory\>/vunit/vhdl/verification_components and then do `python run.py vunit_lib.tb_bus_master_pkg*`. Does it work?

apopa, 2022-12-12:
Hi, thanks for update. I will do it today and let you know about the results. Thanks.

closed 2024-12-11 by LarsAsplund