VUnit/vunit #878: Mixed language vhdl / verilog

petersencurt, 2022-12-01:
I am using questasim to verify a mixed language vhdl/verilog design. I'm using vhdl configurations to run the same testcases on different architectures of the same entity. This has worked so far but I've just introduced a new verilog file in my hierarchy that looks like this:
Testbench (vhdl)
...wrapper (vhdl)
......new_dut (verilog)
......... Module_a (vhdl)

My vhdl configuration no longer works because questasim errors when I try to bind module_a to architecture module_a2 since it is underneath a verilog module now. Any advice how to do this? I realize this may be more of a vhdl question but I'd prefer a solution that works with vunit!

I tried a workaround where I compiled each of module_a1, _a2, etc. Vhdl into separate libraries. I provided a single "library module_an;" statement to incomplete vhdl configurations that stop at the verilog binding, hoping questasim would search the specified lib first but, no luck. It seems questasim just always chooses to bind the first compiled architecture .

I think above could work if I had control over which -L libraries were passed to the vsim command line. For example I could run testbench_a1 compiled into project_lib linking library -L module_a1_lib but not module_a2_lib. Is this supported? If not is it feasible and useful or am I trying to use vunit in ways it was never intended? Perhaps the list of libraries could be specified as part of the vunit configuration?

Thanks for the advice

LarsAsplund, 2022-12-01:
I haven't encountered this specific problem so I'm not sure what the problem is but it looks like the type of problem that you need to work around. Generally speaking I do not have problems with VHDL configurations since I'm not using them. Rather I use generics combined with if-generate statements and *VUnit* configurations to select what code to run.

The reason for using this approach is that it removes some of the limitations with VHDL configurations. We have taken some steps to deal with those limitations but that is still on a development branch. That branch also contain a draft blog on the topic that I recommend that you read for a deeper dive into the topic: https://github.com/VUnit/vunit/blob/add-configuration-support/docs/blog/vhdl_configurations.rst

petersencurt, 2022-12-01:
Thanks for the reply and the link to the blog post. I agree with the blog post:

_This highlights a problem with configurations. All configuration combinations have to be coded manually which doesn’t scale well as more configuration options are added. What is needed is a way to do this programatically_

However I also don't like how the generic if/else generate scales. Once there are more than 2 or 3 different architectures to support making a simple change to the shared entity can require too many changes. It is hard to browse a file when the shared entity has a lot of ports.

I wish there were some way to drive the configuration selection through the hierarchy using something like a generic. It seems to be me that would be the best of both!

LarsAsplund, 2022-12-02:
There has been discussion to add support for providing the architecture of an entity substantiation as a string. That string can then be assigned with a generic. See https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/235

closed 2022-12-04 by LarsAsplund