VUnit/vunit #880: Question: Filtering testbenches, and precompiling specific files

Bonusbartus, 2022-12-07:
After looking through all documentation I still cannot find if/how it is possible to change the settings for finding testbenches.
when not finding any testbenches VUnit gives the following output, suggesting that it is possible to change the filter: 
```
WARNING - Found no test benches using current filter rule:

    Filters entities and modules that have a runner_cfg generic/parameter

    Gives warning when design_unit matches tb_* or *_tb without having a runner_cfg
    Gives warning when a design_unit has  a runner_cfg but the name does not match tb_* or *_tb
```

I my case I want to do (either of) two things:
- Change the regext to scan for something like *_vunit_tb
- NOT give a warning when a tb is found without a runner_cfg
is this possible? is there a python interface to change the filter or the warning?

also is it somehow possible to compile specific files into a specific library manually? I am having problems with one .v file needed for multiple (vhdl) design units, that isn't compiled when trying to run vunit with the "-m" switch.

background: I am currently in the process of setting up a regression testing environment of an existing project, and I don't want to touch the original testbenches. therefore I am writing wrappers around them with the vunit integration. So just instantiate the original testbench + a test_runner process with the runner_config.

Thanks for all the work!

Bonusbartus, 2022-12-07:
Thanks, that works for me 👍 
it wasn't fully clear to me if there was an API available or not!

LarsAsplund, 2022-12-07:
No, it's not obvious. I'll reopen this as I think we should add a proper API.

Bonusbartus, 2022-12-08:
Thanks!
How about the other part of the question: is it possible to manually compile specific files through vunit?
I now solved it by having 2 vunit run.py scripts, one with only specific libraries which contain files that need to be compiled always, and does not contain testbenches, a second one containing all files

LarsAsplund, 2022-12-08:
If the `-m` option fails to compile a file it could mean that VUnit didn't pick up some dependency. What you can do is something like this to add a dependency manually.

``` python
vhdl_file = my_lib.get_source_file(path_to_vhdl_file) # file:///C:/github/vunit/docsbuild20221208/py/vunit.html#vunit.ui.library.Library.get_source_file
verilog_file = my_lib.get_source_file(path_to_verilog_file)

vhdl_file.add_dependency_on(verilog_file) # file:///C:/github/vunit/docsbuild20221208/py/vunit.html#vunit.ui.source.SourceFile.add_dependency_on