VUnit/vunit #881: Improve changelog

GlenNicholls, 2022-12-21:
I've been finding it somewhat painful to read the changelog as it contains a lot of dev-only information. I am fine with this info sitting in a misc section, but it would be nice to have breaking and features sections to make it easier to see what is actually different from a user perspective from one release to the next.

In 4.6 for example, I've highlighted a few entries that are irrelevant to most people:

```
Add Python 3.9 and 3.10 to classifiers.

-> Use MAJOR and MINOR constants to check supported Python version. [#724](https://github.com/VUnit/vunit/issues/724/)

-> Fix pylint issues.

-> Use f-strings for string formatting. [#743](https://github.com/VUnit/vunit/issues/743/) [#747](https://github.com/VUnit/vunit/issues/747/)

-> Specify encoding when using ‘open’. [#748](https://github.com/VUnit/vunit/issues/748/)

-> Set black line-length to 120 characters. [#736](https://github.com/VUnit/vunit/issues/736/)

-> Use Path from pathlib, instead of open().

Add support for log location based on VHDL-2019 call paths. [#729](https://github.com/VUnit/vunit/issues/729/)

GHDL supports VHDL package generics. [#753](https://github.com/VUnit/vunit/issues/753/)

Bump OSVVM to 2021.09.

-> [Tox] Use pytest for collecting coverage, add py310.
...
```

`Fix pylint issues` for example is something I don't find useful, that's a commit message that clutters the relevant release information in a changelog.

I've been using [towncrier](https://towncrier.readthedocs.io/en/stable/) which automates the annoyingness of a changelog. If this is of interest, I'm happy to submit a PR, but Tox and some other projects that use it are good examples as well.

LarsAsplund, 2023-01-05:
Yes, we haven't put any effort into this but from previous experiences with inhouse solutions I prefer a solution where the clarity of the release notes is based on clarity of commit messages and pull requests. To achieve that we should work more on templates. Please go ahead with a PR on this with some examples.

GlenNicholls, 2023-01-05:
One question I have is can we remove `Commits since previous release` in the release notes? I see that more as a tool for developers, but I only really see value in that when there's a regression. In that case, a user is likely just going to open an issue instead of digging through commits to find the cause.

GlenNicholls, 2023-01-06:
PR is ready to be reviewed and has examples attached.