VUnit/vunit #882: All tests fail on Windows with GHDL

Jitterreduction, 2023-01-02:
All testcases fail on my Windows machine which run on Linux perfectly fine. Even the minimal test examples ends with a fail.

`library vunit_lib;
context vunit_lib.vunit_context;

entity tb_minimal is
  generic (runner_cfg : string := runner_cfg_default);
end entity;

architecture tb of tb_minimal is
begin
  test_runner : process
  begin
    test_runner_setup(runner, runner_cfg);
    check_equal(to_string(17), "17");
    test_runner_cleanup(runner);
  end process;
end architecture;
`

Here is the output. The default post_run callback was added.

`
Re-compile not needed

Starting lib.tb_minimal.all
Output file: C:\Users\smu\FPGA\vunit\vunit_out\test_output\lib.tb_minimal.all_42aa262c7c96c708ab3f3960f033f2328c642136\output.txt
simulation stopped @0ms with status 0
fail (P=0 S=0 F=1 T=1) lib.tb_minimal.all (1.3 seconds)

==== Summary ==============================
fail lib.tb_minimal.all (1.3 seconds)
===========================================
pass 0 of 1
fail 1 of 1
===========================================
Total time was 1.3 seconds
Elapsed time was 1.3 seconds
===========================================
Some failed!
C:\Users\smu\FPGA\vunit\vunit_out
lib.tb_minimal.all
failed
1.290325403213501
C:\Users\smu\FPGA\vunit\vunit_out\test_output\lib.tb_minimal.all_42aa262c7c96c708ab3f3960f033f2328c642136
lib.tb_minimal.all_42aa262c7c96c708ab3f3960f033f2328c642136
`

vunit-hdl 4.6.0 with GHDL 3.0.0-dev (2.0.0.r987.gd45f4a31e) is installed and I tested it with python 3.6.8 as well as python 3.8.0.
I have to check the versions used on Linux again, but they are quite similar.

LarsAsplund, 2023-01-02:
@Jitterreduction What is the default `post_run` you're referring to? The default is no post run callback, see https://vunit.github.io/py/vunit.html#vunit.ui.VUnit.main. Can you run with `--log-level=debug` and provide the output?

Jitterreduction, 2023-01-04:
Hi @LarsAsplund,
yeah sorry, the `post_run` was a bit unclear. I was referring to the example in the interface documentation https://vunit.github.io/py/vunit.html#vunit.ui.results.TestResult .

So just to clarification here is the whole script:
```
#!/usr/bin/env python3

from pathlib import Path
from vunit import VUnit

def post_func(results):
    report = results.get_report()
    print(report.output_path)
    for key, test in report.tests.items():
        print(key)
        print(test.status)
        print(test.time)
        print(test.path)
        print(test.relpath)

VU = VUnit.from_argv()

lib = VU.add_library("lib")
lib.add_source_files(Path(__file__).parent / "*.vhd")

VU.main(post_run=post_func)

```

And yeah, here is the output with the debug level flag. Sadly, there seems to be still no clue about what exaclty is causing the issue.

```
$ python vunit_run.py  --log-level=debug
  DEBUG - Adding library vunit_lib with path C:\Users\smu\FPGA\vunit\vunit_out\ghdl\libraries\vunit_lib
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\byte_vector_ptr_pkg.vhd to library vunit_lib
  DEBUG - Adding primary design unit (package) byte_vector_ptr_pkg
  DEBUG - Adding secondary design unit (package body) for package byte_vector_ptr_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\byte_vector_ptr_pkg.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec-2008p.vhd to library vunit_lib
  DEBUG - Adding primary design unit (package) codec_2008p_pkg
  DEBUG - Adding secondary design unit (package body) for package codec_2008p_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec-2008p.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec.vhd to library vunit_lib
  DEBUG - Adding primary design unit (package) codec_pkg
  DEBUG - Adding secondary design unit (package body) for package codec_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec_builder-2008p.vhd to library vunit_lib
  DEBUG - Adding primary design unit (package) codec_builder_2008p_pkg
  DEBUG - Adding secondary design unit (package body) for package codec_builder_2008p_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec_builder-2008p.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec_builder.vhd to library vunit_lib
  DEBUG - Adding primary design unit (package) codec_builder_pkg
  DEBUG - Adding secondary design unit (package body) for package codec_builder_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\codec_builder.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\data_types_context.vhd to library vunit_lib
  DEBUG - Adding primary design unit (context) data_types_context
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\data_types_context.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\dict_pkg.vhd to library vunit_lib
  DEBUG - Adding primary design unit (package) dict_pkg
  DEBUG - Adding secondary design unit (package body) for package dict_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\dict_pkg.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\integer_array_pkg-body.vhd to library vunit_lib
  DEBUG - Adding secondary design unit (package body) for package integer_array_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\integer_array_pkg-body.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\integer_array_pkg.vhd to library vunit_lib
  DEBUG - Adding primary design unit (package) integer_array_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\integer_array_pkg.vhd' has no components
  DEBUG - Adding source file C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\integer_vector_ptr_pkg-body-2002p.vhd to library vunit_lib
  DEBUG - Adding secondary design unit (package body) for package integer_vector_ptr_pkg
  DEBUG - The file 'C:\Users\smu\AppData\Roaming\Python\Python36\site-packages\vunit\vhdl\data_types\src\integer_vector_ptr_pkg-body-2002p.vhd' has no components
[... 376 lines cut ...]
simulation stopped @0ms with status 0
  DEBUG - TestRunner: Leaving
fail (P=0 S=0 F=1 T=1) lib.tb_minimal.all (1.1 seconds)

==== Summary ==============================
fail lib.tb_minimal.all (1.1 seconds)
===========================================
pass 0 of 1
[... cut ...]
Total time was 1.1 seconds
Elapsed time was 1.1 seconds
===========================================
Some failed!
C:\Users\smu\FPGA\vunit\vunit_out
lib.tb_minimal.all
failed
1.1304395198822021
C:\Users\smu\FPGA\vunit\vunit_out\test_output\lib.tb_minimal.all_42aa262c7c96c708ab3f3960f033f2328c642136
lib.tb_minimal.all_42aa262c7c96c708ab3f3960f033f2328c642136

```

Here is also the whole report with recompiling the simulation
[debug_with_clean.txt](https://github.com/VUnit/vunit/files/10342628/debug_with_clean.txt)

LarsAsplund, 2023-01-05:
The return code (0xc0000005) is from GHDL. I don't know what it means though. As a first step I would revert to an older, more stable version of GHDL. If that makes the error go away, I would bring the issue to GHDL.

Jitterreduction, 2023-01-05:
I'd love to give it a try with an older version, but only the nightly build is available as standalone. And while building it on Linux is an easy tasks for me, I struggle with it on Windows.

LarsAsplund, 2023-01-06:
@Jitterreduction I updated my GHDL to the latest build and see the same problem. I opened a bug report at GHDL: https://github.com/ghdl/ghdl/issues/2302

I've attached version:

> GHDL 3.0.0-dev (2.0.0.r600.gcdcbb258b) [Dunoon edition]
> Compiled with GNAT Version: 12.1.0
> mcode code generator

which works on my side.

[ghdl.zip](https://github.com/VUnit/vunit/files/10363107/ghdl.zip)

Jitterreduction, 2023-01-09:
@LarsAsplund , thank you for the uploaded ghdl. It also works on my side :)

umarcor, 2023-02-28:
I just tested all the acceptance and vcomponents tests on mcode MINGW32, and everything is green:

- https://github.com/VUnit/vunit/actions/runs/4287938039/jobs/7469362087
- https://github.com/VUnit/vunit/actions/runs/4287938039/jobs/7469362331

@Jitterreduction maybe you were mistakenly trying mcode for 64 bits on Windows? That's unsupported. You need to use LLVM on Windows 64 bits. In order to use mcode on Windows for simulation purposes, it needs to be 32 bits.

Jitterreduction, 2023-03-07:
@umarcor , thank you very much. This happened in my case. When I did the test on Windows I used the MINGW64-mcode-standalone. With the current ghdl-MINGW32, every is working fine.

@LarsAsplund , have you also made the same mistake? If not, then there might have been a bug, which isn't present anymore.

closed 2023-03-07 by eine