VUnit/vunit #886: 2 tests fail

yurivict, 2023-01-08:
```
========================================================================================== test session starts ==========================================================================================
platform freebsd13 -- Python 3.9.16, pytest-7.2.0, pluggy-1.0.0 -- /usr/local/bin/python3.9
cachedir: .pytest_cache
hypothesis profile 'default' -> database=DirectoryBasedExampleDatabase('/usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/.hypothesis/examples')
Using --randomly-seed=1703764193
rootdir: /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0
plugins: forked-1.4.0, hypothesis-6.58.0, cov-2.9.0, randomly-3.12.0, timeout-2.1.0, rerunfailures-10.1, flaky-3.7.0, xdist-2.5.0, env-0.6.2, freezegun-0.4.2, mock-1.10.4
collected 509 items / 2 errors                                                                                                                                                                          

================================================================================================ ERRORS =================================================================================================
_________________________________________________________________________ ERROR collecting tests/acceptance/test_artificial.py __________________________________________________________________________
tests/acceptance/test_artificial.py:23: in <module>
    class TestVunitArtificial(unittest.TestCase):
tests/acceptance/test_artificial.py:41: in TestVunitArtificial
    simulator_is("modelsim", "rivierapro"),
vunit/sim_if/common.py:25: in simulator_is
    return SIMULATOR_FACTORY.select_simulator().name in names
E   AttributeError: 'NoneType' object has no attribute 'name'
____________________________________________________________________ ERROR collecting tests/acceptance/test_external_run_scripts.py _____________________________________________________________________
tests/acceptance/test_external_run_scripts.py:34: in <module>
    class TestExternalRunScripts(TestCase):
tests/acceptance/test_external_run_scripts.py:39: in TestExternalRunScripts
    @mark.skipif(not simulator_supports_verilog(), reason="Requires a Verilog simulator")
tests/acceptance/test_external_run_scripts.py:29: in simulator_supports_verilog
    return simulator_is("modelsim", "incisive")
vunit/sim_if/common.py:25: in simulator_is
    return SIMULATOR_FACTORY.select_simulator().name in names
E   AttributeError: 'NoneType' object has no attribute 'name'
=========================================================================================== warnings summary ============================================================================================
vunit/test/bench.py:29
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/bench.py:29: PytestCollectionWarning: cannot collect test class 'TestBench' because it has a __init__ constructor (from: tests/unit/test_test_bench.py)
    class TestBench(ConfigurationVisitor):

vunit/test/bench.py:276
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/bench.py:276: PytestCollectionWarning: cannot collect test class 'Test' because it has a __init__ constructor (from: tests/unit/test_test_bench.py)
    class Test(object):

vunit/test/bench_list.py:20
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/bench_list.py:20: PytestCollectionWarning: cannot collect test class 'TestBenchList' because it has a __init__ constructor (from: tests/unit/test_test_bench_list.py)
    class TestBenchList(object):

vunit/test/report.py:21
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/report.py:21: PytestCollectionWarning: cannot collect test class 'TestReport' because it has a __init__ constructor (from: tests/unit/test_test_report.py)
    class TestReport(object):

vunit/test/runner.py:27
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/runner.py:27: PytestCollectionWarning: cannot collect test class 'TestRunner' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestRunner(object):  # pylint: disable=too-many-instance-attributes

vunit/test/report.py:21
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/report.py:21: PytestCollectionWarning: cannot collect test class 'TestReport' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestReport(object):

vunit/test/list.py:14
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/list.py:14: PytestCollectionWarning: cannot collect test class 'TestList' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestList(object):

tests/unit/test_test_runner.py:217
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/tests/unit/test_test_runner.py:217: PytestCollectionWarning: cannot collect test class 'TestCaseMock' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestCaseMock(object):

vunit/test/suites.py:147
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/vunit/test/suites.py:147: PytestCollectionWarning: cannot collect test class 'TestRun' because it has a __init__ constructor (from: tests/unit/test_test_suites.py)
    class TestRun(object):

tests/unit/test_ui.py:1218
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit_hdl-4.6.0/tests/unit/test_ui.py:1218: PytestCollectionWarning: cannot collect test class 'TestPreprocessor' because it has a __init__ constructor (from: tests/unit/test_ui.py)
    class TestPreprocessor(object):

../../../../../local/lib/python3.9/site-packages/pytest_freezegun.py:17: 1018 warnings
  /usr/local/lib/python3.9/site-packages/pytest_freezegun.py:17: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
    if LooseVersion(pytest.__version__) < LooseVersion('3.6.0'):

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
=================================================================================== 1028 warnings, 2 errors in 2.17s ====================================================================================
*** Error code 2
```

Version: 4.6.0
Python: 3.9

LarsAsplund, 2023-01-10:
This is a bad error message but it looks like it fails to find a simulator for running tests. Are you familiar with https://vunit.github.io/cli.html#simulator-selection?

LarsAsplund, 2023-01-23:
@yurivict Did you manage to confirm that the lack of simulator is the cause of this?

yurivict, 2023-01-23:
@LarsAsplund Thank you for this info. I will try to make ghdl running.

closed 2023-01-23 by yurivict