VUnit/vunit #893: Break in subprogram stop at run.vhd lien 132

RomoloMurri97, 2023-01-20:
Hello, I am running a VUnit testbench but it seams to break it instead of finishing

LarsAsplund, 2023-01-20:
What version are you running? Alternatively you can show me what line 132 is.

RomoloMurri97, 2023-01-20:
vunit/4.5.0, thank you for answering. Anyway:
[image]

LarsAsplund, 2023-01-20:
Ok, that is an odd line to crash (?) on. You would need to provide the output of your simulation, error messages etc. What simulator do you use and does this happen for all testbenches or only one. If you can provide the actual code you're running or a simple example with the same behavior.

RomoloMurri97, 2023-01-23:
I need to cover some part of the code, but this is the output
[image]
Basically what happens if it doesn't break is that it just continues to the infinite. So what makes the simulation actually stops?
By taking a look into run.vhd it seems that there is something wrong in the lines right before 131, so:
    if has_active_python_runner(runner_state) then
      core_pkg.test_suite_done;
    end if;

Anyway this is the code that I run:
[image]

LarsAsplund, 2023-01-23:
What if you run 

``` python
python <vunit install dir>/examples/vhdl/user_guide/run.py *.all -v
```
Is the output different?

I normally do not run with Questa but making a test with Questa 2022.4 gives an output looking like this

[image]

Your simulation is showing a break just before getting to that line but I suspect this may have to do with optimization. The mapping of where it stopped in the optimized code to the correct source code line may not be optimal.

If `core_pkg.test_suite_done;` isn't performed correctly then the test would be failed at Python level and you wouldn't get a pass. `p_simulation_exit_is_disabled(runner_state)` is a backdoor used for verification of VUnit. It will always return false in a normal simulation which means that `core_pkg.stop(0);` will be called. That procedure simply breaks the simulation. This is what we see but it looks like Questa is pointing at the wrong line.

RomoloMurri97, 2023-01-23:
I just tried and apparently, it gives the same output, so breaks at that lines. Is it something I should really care about in your opinion?

LarsAsplund, 2023-01-23:
No, I wouldn't be concerned about this.

closed 2023-01-24 by LarsAsplund