VUnit/vunit #898: GHDLInterface can not detect LLVM backend with newer GHDL/LLVM

LukasVik, 2023-02-14:
`GHDLInterface.determine_backend` fails because `ghdl --version` reports 

```
...
llvm 14.0.0 code generator
...
```

Not just `llvm code generator` as `GHDLInterface.determine_backend` expects, which I assume is how it used to be. Should probably update that method to check with a regexp. 

I am busy the upcoming days, but can make a PR within a week. Unless anyone else wants to fix it before that.

Full output example (from gitter):

```
me@testVM:~\\testprj$ python run.py \*test
ERROR - Could not detect known LLVM backend by parsing 'ghdl --version' 
Expected to find one of dict\_keys(\['mcode code generator', 'llvm code generator', 'GCC back-end code generator'\])
== Output of 'ghdl --version'============================================================
GHDL 3.0.0-dev (2.0.0.r1367.g803edb716) \[Dunoon edition\]
Compiled with GNAT Version: 10.4.0
llvm 14.0.0 code generator
Written by Tristan Gingold.
Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. 
=========================================================================================
```

LarsAsplund, 2023-02-14:
@LukasVik Try now.

LukasVik, 2023-02-15:
Works wonderfully, thanks!

closed 2023-02-15 by LukasVik