VUnit/vunit #899: How to apply the "3-step flow" using Questa

jfrensch, 2023-02-16:
Hi all,

my understanding is, that the VUnit framework (always?) uses the "2-step flow" (-> vcom & vsim) of Questa, where the 'vopt' step is automatically applied during 'vsim'.
Unfortunaly, I need some intermediate result of the vopt step to be able to use the 'Visuallizer' to analyze the results of a simulation:

      "vopt -debug -designfile design.bin -o tb_opt tb" 
not only provides an optimized design 'tb_opt' of the original testbench 'tb' for faster simulation, but also a database (-> design.bin) required by the 'Visualizer' to correlate the simulation results from 'tb_opt' to the design being simulated.

How can/should I apply that 3rd step (vopt) within the VUnit framework?

Many thanks for advice
Jochen

tasgomes, 2023-08-02:
This 3-step flow would also solve this issue https://github.com/VUnit/vunit/issues/877

LarsAsplund, 2024-04-08:
Good news today is that Siemens decided to support VUnit with Questa licenses such that issues like this be solved.

LarsAsplund, 2024-06-27:
I started to prototype on this and there is a first iteration to try out where I simply run `vopt` before calling `vsim`. I'm only using a few options in the `vopt` call, most importantly is probably `-floatgenerics` which is applied recursively on the simulation top-level. The `runner_cfg` generic needs to be floating since it has no default value. Is there a need to control that more selectively to enable optimization for any custom generics added to the testbench?

In this iteration, optimization is not a proper step before the simulation step. This means that `vopt` is called before starting the simulation of **each** test in a testbench. This is a problem as noted in #877.

```
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vopt-6) -- Waiting for lock by "larsa@LAPTOP-B6KUINVE, pid = 25780
# ". Lockfile is "C:/github/vunit/examples/vhdl/three_step_flow/vunit_out/modelsim/libraries/lib/_lock".
# ** Error: (vopt-2261) 'lib.opt_libtbexampletb' is already an optimized design.
# Optimization failed
```

This error is not suppressible so this doesn't help (I've added support for setting `vopt` options, both in batch mode and in gui mode)

`vu.set_sim_option("modelsim.vopt_flags", ["-suppress", "2261"])`

If I run the tests after one another this doesn't seem to be a concern. `vopt` is simply skipped if the design is already optimized:

`# Incremental compilation check found no design-units have changed.`

Making optimization a proper step executed before starting the simulation step is probably the solution to this.

Until then, please give it a try for your other use cases.

LarsAsplund, 2024-06-27:
Btw, there is a small example that you can start playing with in https://github.com/VUnit/vunit/tree/three-step-flow/examples/vhdl/three_step_flow

xkvkraoSICKAG, 2024-08-16:
I tried the three-step-flow branch and it will cause errors when running with multiple threads. The vopt call will need to be thread safe, you cannot run vopt in parallel with the same output destination apparently. 

It can be solved either by:
1. Adding a thread unique suffix to each vopt output destination
2. Calling vopt call directly from Python protected by a mutex/lock. There would be dictionary of locks with the top level name as the key so that vopt of a given top level test bench is only done once and not in parallel with simulation.

The benefit of 2 is that it skips doing vopt for every simulation which could reduce some overhead.

xkvkraoSICKAG, 2024-08-16:
I have made a prototype of solution 1. described above and it works without problems in multi threading. This is really the simplest solution to get it working as separating the vopt step and using a Python-lock on it would require a lot of restructuring of `modelsim.py` vs `vsim_simulator_mixin.py`.

xkvkraoSICKAG, 2024-08-16:
Note also that I found problems with the `floatgenerics` argument. It caused vopt to hang indefinitely on some of my test benches. By removing the dot from the end of `-floatgenerics+top.` to `-floatgenerics+top` it no longer hanged. 

My understanding from the manual is that the trailing dot causes the generics of all lower instances to also be floating. For the purpose of VUnit it should be enough that the top level test bench generics are floating as changing the generic of a deeper instance is not required or supported. I would assume a floating top level generic coupled to a lower instance generic would also cause it to be floating anyway without the trailing dot.

xkvkraoSICKAG, 2024-08-16:
It seems running vopt will mutate the common files in the library folder even if multiple threads use different vopt output targets. I have verified this by diffing all md5sum of all files in the library folder before and after running vopt. However just running vsim on an already created vopt folder does not seem to change any md5sum at all. 

This makes me think a solution needs to ensure all vopt calls for a single library needs to happen before any simulation starts.
Even between two test benches within the same library the vopt calls cannot run in parallel it seems.

LarsAsplund, 2024-08-19:
@xkvkraoSICKAG Thanks for trying this out.

Yes, adding a thread suffix to the name of the optimized design would be a nice solution. It works in my simple example but I also see that the library files are modified.

As I see it, duplication is the only option. I will give it a try. I will also consult Siemens to get these observations verified.

The reason for using floating generics on all levels is that there are use cases where the `runner_cfg` are passed to a lower-level entity. However, that is a less common use case  that we can ignore in the first iterations.

LarsAsplund, 2024-08-19:
@xkvkraoSICKAG Different directories for each thread is something that was already implemented in another feature branch so there is code to reuse from that branch (which was dropped when we realized that there were other ways to solve that feature).

Library duplication will have to be added and I've a discussion ongoing with Siemens to figure out to what extent that is needed. That will be an overhead that we obviously want to minimize so it will probably only be activated with the 3-step flow. The 2-step flow will work as before.

xkvkraoSICKAG, 2024-08-20:
I have another observation to report. I tried changing the library folder format from `vlib -type flat` which is the default to `vlib -type directory`. This annoyingly *almost* became thread safe but unfortunately Questa still makes a tiny modification to the `_info` file in the library folder when running vopt.

Regarding library duplication. To reduce the overhead it could maybe use a smart approach based on https://docs.python.org/3/library/filecmp.html to only copy over what has changed.

Another approach would be to run `vopt` as part of the compile step of VUnit which is already running in single threaded mode.
This would involve refactoring to also extract more information from the dependency scanner to be able to incrementally know which top level test benches need to be re-optimized.

LarsAsplund, 2024-08-20:
I also tried with `-type directory` and found that the libs are modified if I'm running tests on the same testbench that differ in the top-level generics. It looks like there are modifications to take into account beyond the small `_info` change. I will start with the assumption that everything needs to be copied. That can always be improved if Siemens can provide some valuable insights into the details of the `vopt` behavior.

xkvkraoSICKAG, 2024-08-21:
I pushed my local changes I used to test to a fork:
https://github.com/xkvkraoSICKAG/vunit/commits/three-step-flow2/

I think this commit may be of interest, it ensures the library mapping arguments are deterministic between calls to vcom/vlog and vopt. Before this change they were subject to the random iteration order of dictionary keys:
https://github.com/xkvkraoSICKAG/vunit/commit/be872712a0910f5d4a68822a4aa93e9b4eec910c

LarsAsplund, 2024-08-21:
@xkvkraoSICKAG Can you make a pull request to the VUnit repo?

xkvkraoSICKAG, 2024-08-23:
@LarsAsplund Yes if you can rebase the `three-step-flow` branch on VUnit master I can make a PR to `three-step-flow`.

LarsAsplund, 2024-08-31:
I started to build on a solution where the first test running a testbench performs the optmization which is then used by the other tests. The optimization is still part of the simulation step so the second test has to wait for both the optimization and simulation of the first test to complete before it can proceed. That will be fixed later but I did see something that needs more investigation

Below is a debug log from a testbench with an single test that has 5 different configurations. I'm using two threads for my test run.

The first test run gets to optimize the testbench:

```
2024-08-31 19:20:04,426 - DEBUG - (lib.tb_example.0.test) Optimizing lib.tb_example(tb)
```

Since the second test starts simultaneously in another thread, it blocks while waiting for the first test:

```
2024-08-31 19:20:04,426 - DEBUG - (lib.tb_example.1.test) Waiting for lib.tb_example(tb) to be optimized.
2024-08-31 19:20:04,429 - DEBUG - Starting lib.tb_example.0.test simulation
2024-08-31 19:20:07,808 - DEBUG - lib.tb_example.0.test simulation completed
2024-08-31 19:20:07,808 - DEBUG - lib.tb_example(tb) optimization completed
```

Now the second test case can proceed:

```
2024-08-31 19:20:07,811 - DEBUG - Starting lib.tb_example.1.test simulation
```

With the first test completed, there is one simulation thread available and the third test
can start. At this point there is no need to wait for the optimized testbench:

```
2024-08-31 19:20:07,826 - DEBUG - (lib.tb_example.2.test) Reusing optimized lib.tb_example(tb)
2024-08-31 19:20:07,829 - DEBUG - Starting lib.tb_example.2.test simulation
```

For every test completed, a new one can start:

```
2024-08-31 19:20:10,670 - DEBUG - lib.tb_example.1.test simulation completed
2024-08-31 19:20:10,687 - DEBUG - (lib.tb_example.3.test) Reusing optimized lib.tb_example(tb)
2024-08-31 19:20:10,691 - DEBUG - Starting lib.tb_example.3.test simulation
2024-08-31 19:20:13,107 - DEBUG - lib.tb_example.3.test simulation completed
2024-08-31 19:20:13,127 - DEBUG - (lib.tb_example.4.test) Reusing optimized lib.tb_example(tb)
2024-08-31 19:20:13,130 - DEBUG - Starting lib.tb_example.4.test simulation
2024-08-31 19:20:15,640 - DEBUG - lib.tb_example.4.test simulation completed
```

But what happened to the third test case? It takes forever to complete and is overtaken by the tests starting after it.

```
2024-08-31 19:20:27,080 - DEBUG - lib.tb_example.2.test simulation completed
```

Regardless how many configurations I create of the test, there is always one which completes much
later than the others. That is something I have to investigate further.

LarsAsplund, 2024-09-01:
It should be said that a single-thread test run works as expected. The first test takes a bit longer to run since it's doing the optimization:

[image]

Some test runs with two threads work well. In this case, the second test also takes some extra time since it's waiting for the first to complete. After that everything runs smoothly:

[image]

This is what a bad run with two threads looks like:

[image]

LarsAsplund, 2024-09-01:
What I see is that it is the simulation process that takes time and the problem is intermittent. This is the execution time for 500 configurations of the same test:

[image]

Considering that I once got this message, I'm suspecting this has to do with the license server. In my case it sits on my computer so there is no network delay.

[image]

After trying for 30 seconds it simply fails. I will cleanup my code so that you can test on your computers.

xkvkraoSICKAG, 2024-09-02:
> I started to build on a solution where the first test running a testbench performs the optmization which is then used by the other tests. The optimization is still part of the simulation step so the [...]

Based on my investigations running vopt on any design in a top level will mutate the library_folder/_info file. So running vopt in a library has to lock the entire library. Thus care has to be taken if there are several test benches in the same library, they cannot have vopt run in parallel.

LarsAsplund, 2024-09-02:
Agree, there will be multiple conditions for when vsim and vopt can be run. vsim waits for the testbench to be optimized if it hasn't already and vopt waits for the lib to be available. I hope that a second vopt on a library doesn't invalidate previous vopts on that library just because of the altered _info file.

xkvkraoSICKAG, 2024-09-02:
> Agree, there will be multiple conditions for when vsim and vopt can be run. vsim waits for the testbench to be optimized if it hasn't already and vopt waits for the lib to be available. I hope that [...]

Unfortunately I think the altered _info file does cause problems. I am running our company internal simulations on my three-step-flow branch. On this branch only the _info file is mutated during vopt and still it causes test cases to fail with a low probability.  To mitigate this every test bench in the same library must be sequentially vopt:ed which kind of defeats the common vunit style of having multiple test benches in the same library.

LarsAsplund, 2024-09-03:
I was thinking about the case where you vopt every testbench sequentially. If you vopt A and then B, will you then have to vopt A again before running just because _info changed? Even if the designs didn't change?

The vopt lock would be a problem if you have testbenches without test cases. They would run in series. If you have test cases, only the first test's vopt will run on it's own. All the vsims will run in parallel. If scheduling  is optimised, the vopt for the next testbench could run in parallel with the vsims of the previous.

xkvkraoSICKAG, 2024-09-03:
> I was thinking about the case where you vopt every testbench sequentially. 

Yes the problem is the _info mutation forces you to run vopt sequentially for all testbenches within a library before starting any simulation. This becomes a problem if you have a lot of test benches within the same library.

LarsAsplund, 2024-09-03:
@xkvkraoSICKAG  Ok, so what I have now is a prototype that manages locks for the libraries as well. There are corners which I have yet to handle but it should be useful for testing in some different projects. I've tested it for myself with dummy testbenches and two threads and also with a client that has a single license. What I found was that using two threads improved performance even if there was only one license. Not sure why but maybe vopt is allowed to run concurrently with vsim on a single licens.

Running vopt on one testbench in one thread while running vsim on a testbench already optimized in another thread hasn't caused any problems for me. This is when I run with two licenses. Running two vopts at the same time on different libraries also works.

If you have a library with 10 testbenches and simulation time is much longer than optimization time you will eventually have 10 simulations running concurrently, provided you have that many licenses. The problem is if simulation time is relatively small compared to the optimization time. But is optimization needed in those cases?

I will push what I have tomorrow so you can try it.

LarsAsplund, 2024-09-04:
@xkvkraoSICKAG I pushed an update now. Can you test it with your real-life project? I still see that some tests takes a long time so it would be interesting to see what you experience. I've kept the recursive option on `floatgenerics` so feel free to test without if you run into problems. You can also run with the `--log-level=debug` option to get som info on how the threads synchronize with respect to each other.

tasgomes, 2024-09-05:
@LarsAsplund I did some tests using commit https://github.com/VUnit/vunit/commit/1ce6bbadd41728b7b24c27ef2d67d7fe099ef1c6.

I still end up with problems when running 5 test cases with 5 threads.
4 tests belong to the same library and 1 test belongs to another.

[questa_error.txt](https://github.com/user-attachments/files/16893434/questa_error.txt)

Sometimes it does run without issues but it is really random... I can have multiple runs without issues and then have 2 consecutive runs with issues.

LarsAsplund, 2024-09-05:
@tasgomes Can you start by running the example here: https://github.com/VUnit/vunit/tree/three-step-flow/examples/vhdl/three_step_flow?

How did you enable optimization, see https://github.com/VUnit/vunit/blob/1ce6bbadd41728b7b24c27ef2d67d7fe099ef1c6/examples/vhdl/three_step_flow/run.py#L31

tasgomes, 2024-09-06:
@LarsAsplund I ran your example, please find the logs below:

Single-thread:
[good.log](https://github.com/user-attachments/files/16906748/good.log)

5-thread:
[error.log](https://github.com/user-attachments/files/16906747/error.log)

LarsAsplund, 2024-09-06:
Looking at the error part:

```
# vsim -modelsimini C:/Git/et-fw/Demos/ExampleProject/lib/FW/VHDL/vunit/examples/vhdl/three_step_flow/vunit_out/modelsim/modelsim.ini -wlf C:/Git/et-fw/Demos/ExampleProject/lib/FW/VHDL/vunit/examples/vhdl/three_step_flow/vunit_out/test_output/lib2.tb_example.054861d12681315f7dce6ece3407eb12387d16940/modelsim/vsim.wlf -work lib2 -quiet -t ps -onfinish stop opt_lib2tbexampletb -L vunit_lib -L lib1 -L lib2 -g/tb_example/runner_cfg="active python runner : true,enabled_test_cases : test,output path : C::/Git/et-fw/Demos/ExampleProject/lib/FW/VHDL/vunit/examples/vhdl/three_step_flow/vunit_out/test_output/lib2.tb_example.054861d12681315f7dce6ece3407eb12387d16940/,tb path : C::/Git/et-fw/Demos/ExampleProject/lib/FW/VHDL/vunit/examples/vhdl/three_step_flow/,use_color : true" -g/tb_example/value=0
# Start time: 12:50:27 on Sep 06,2024
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vopt-6) -- Waiting for lock by "tgomes@BELUGA, pid = 10148
# ". Lockfile is "C:/Git/et-fw/Demos/ExampleProject/lib/FW/VHDL/vunit/examples/vhdl/three_step_flow/vunit_out/modelsim/libraries/lib2/_lock".
# ** Error: (vopt-2261) 'lib2.opt_lib2tbexampletb' is already an optimized design.
# Optimization failed
```

VUnit calls `vsim` with an optimized design (`opt_lib2tbexampletb`) but Questa still thinks it needs to be optimized (`Design is being optimized...`) but eventually it figures out that it is already optimized (`'lib2.opt_lib2tbexampletb' is already an optimized design.`) and fails.

The fact that Questa is waiting for a lock file suggests that there is a race between VUnit's internal thread synchronization and the lock state of Questa. The VUnit threads are synchronized with OS mechanisms. When one `vopt` call returns, VUnit will let another thread to call it regardless of any lock file. I will update to take the lock file into account and see if that helps.

LarsAsplund, 2024-09-06:
Pushed a new version where i check and wait for the lock file to be removed. Since I can't recreate your problem, I haven't been able to test it properly.

I only have two lecenses. Does it work with `-p2`?

tasgomes, 2024-09-09:
@LarsAsplund I still get some errors sometimes, see two examples below with `-p 2`:

[error_2a.log](https://github.com/user-attachments/files/16930841/error_2a.log)
[error_2b.log](https://github.com/user-attachments/files/16930843/error_2b.log)

LarsAsplund, 2024-09-09:
@tasgomes One error is the same as the one we saw before but the other is new. @xkvkraoSICKAG and I concluded that `vopt` on different libraries can be done in parallel since only the `_info` file in the library to which the optimized design belongs is modified. The second error suggests that this is not the case. I pushed a quickfix that prevents two vopts to run at the same time. Let's see if that helps.

tasgomes, 2024-09-11:
Hi @LarsAsplund, it seems stable with 2 threads but if I increase the number of threads then issues appear again. See below an example with 6 threads:

[error_3.log](https://github.com/user-attachments/files/16958721/error_3.log)

tasgomes, 2024-09-11:
As referenced before, it seems like Questa tries to optimize a design that is already optimized, and Questa throws an error in this case:

```
# Start time: 08:14:52 on Sep 11,2024
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vopt-6) -- Waiting for lock by "tgomes@BELUGA, pid = 20312
# ". Lockfile is "C:/Git/et-fw/Demos/ExampleProject/lib/FW/VHDL/vunit/examples/vhdl/three_step_flow/vunit_out/modelsim/libraries/lib2/_lock".
# ** Error: (vopt-2261) 'lib2.opt_lib2tbexampletb' is already an optimized design.
# Optimization failed
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=1, Warnings=1.
# Error loading design
Error loading design
# End time: 08:14:55 on Sep 11,2024, Elapsed time: 0:00:03
# Errors: 1, Warnings: 1
```

Perhaps we could query and check first if the design is already optimized or not and then skip optimization in that case. There is a `vopt` argument that apparently allows to skip optimization:

[image]

I checked Questa documentation and also found the following that could shed some light on how to "lock" optimizations for "multi-thread" cases:

[image]

[image]

LarsAsplund, 2024-09-11:
@tasgomes When we run `vopt`, we create a name for the optimized design that differs from the original design and call `vsim` with that. The optimized design is named as the non-optimized design but with all non-alphanumeric characters removed and an `opt_` prefix. The log tells us that `vsim` is being called with `opt_lib2tbexampletb` so there is no need for an extra query. However, to be on the safe side I suggest that you add this to your run script

```
vu.set_sim_option("modelsim.vsim_flags", ["-novopt", "-suppress", "12110"])
```
The `-novopt` option will result in an error in the normal case when `vsim` correctly recognizes that its design input is already optimized but that error (12110) can be suppressed,

I'm suspecting that the confusion about the state of the design has to do with the lock file `vsim` is complaining about. So far I only prevented several `vopt` calls at the same time but not that simulations are run during `vopt`. It could be that `vsim` can't do that. In the best scenario, `vsim` in one lib can still run concurrently with `vopt` on another lib. To test that theory you could try:

```
python run.py -p2 --log-level=debug lib2.tb_example.* lib1.*
```

Now we only have two testbenches located in separate libs. Once the first testbench has been optimized in one of the libs, the second testbench will be optimized to the other lib concurrently with the simulations of the testbench in the first lib.

If `vopt` can't be executed with any `vsim`, it will be much more limiting. The simplest solution to that would cause a flow where one testbench is optimized first. Then all its test cases will be simulated in parallel before the pattern repeats with the second testbench. If the first testbench has 2 test cases, we will only run 2 parallel threads even if we have licenses and CPU cores that can support more threads.

To get around that we would have to optimize all testbenches first in series and then do all simulations in parallel. That would affect the VUnit design at a higher level as we cannot hide it locally in the Questa/Modelsim interface code anymore.

tasgomes, 2024-09-12:
@LarsAsplund I added:

`vu.set_sim_option("modelsim.vsim_flags", ["-novopt", "-suppress", "12110"])`

Sometimes it works and sometimes fails as shown below:

[good_4.log](https://github.com/user-attachments/files/16974552/good_4.log)
[error_4.log](https://github.com/user-attachments/files/16974519/error_4.log)

Then I try:

`python run.py -p2 --log-level=debug lib2.tb_example.* lib1.*`

I get similar results, either:
- it works
- it fails with `# ** Error: (vsim-3170) Could not find 'opt_lib2tbexampletb'.` when using `-novopt`
- it fails with `# ** Error: (vopt-2261) 'lib2.opt_lib2tbexampletb' is already an optimized design.` without the `-novopt` flag

LarsAsplund, 2024-09-12:
@tasgomes  I suspect that the new "could not find" error message can be just another expression of Questa being confused rather than the file actually being missing. However, to be on the safe side, I will make sure that the output from `vopt` really exists in the file system before proceeding.

The next step is to also disallow `vsim` runs concurrent to `vopt` such that `vopt` always runs in complete isolation.

LarsAsplund, 2024-09-12:
@tasgomes Before doing anything else, can you do `pip install watchdog` and then run with the latest update? It will log changes in the file system. Maybe that can help us understand what is going on.

tasgomes, 2024-09-13:
@LarsAsplund 

A good run for your reference: 
[good_5.log](https://github.com/user-attachments/files/16989506/good_5.log)

A bad run with `-novopt`:
[error_5a.log](https://github.com/user-attachments/files/16989512/error_5a.log)

A bad run without `-novopt`:
[error_5b.log](https://github.com/user-attachments/files/16989514/error_5b.log)

LarsAsplund, 2024-09-15:
@tasgomes Ok, now I see it. There is a bug in my code which causes a race condition. Please try the latest push.

LarsAsplund, 2024-09-16:
I also started to remove the things I added lately before finding what I think was the real bug. I suggest testing the last three commits one at a time to see if removing everything was too optimistic

LarsAsplund, 2024-09-18:
@tasgomes @xkvkraoSICKAG Have any of you had the chance to test the latest commit with your projects?

tasgomes, 2024-09-18:
@LarsAsplund I am out of office this week. I can retry this again next Monday.

tasgomes, 2024-09-23:
@LarsAsplund I am back. Below you find the last three commits one at a time. I ran each several times without problems, except for the last one. This one has an issue that occurs only sometimes.

[00_3a976cd.log](https://github.com/user-attachments/files/17093503/00_3a976cd.log)
[01_28743da.log](https://github.com/user-attachments/files/17093504/01_28743da.log)
[02_7032b5e.log](https://github.com/user-attachments/files/17093505/02_7032b5e.log)

LarsAsplund, 2024-09-23:
@tasgomes Thanks @tasgomes. I suspect there can be a slight delay before files owned by one `vopt` call (lock files or other files) are properly released on the file system. A second `vopt` call, made after the first one returns, may run into that.

However, in this case `vopt` fails on the first call on `lib2`. The only previous call was on `lib1`. There should not be any prior activity on `lib2` files unless:
1. You had a previous run that crashed somehow such that some files were left open and/or lock files were not deleted
2. Our assumption that `vopt` on one library doesn't affect other libraries was wrong. In this case the examples are standalone but what would happen if a testbench in `lib1` uses a component from `lib2`.

I'll reach out to Siemens for some more support. As long as we are guessing how it works, we cannot be sure we have a stable solution

tasgomes, 2024-09-24:
@LarsAsplund I restarted my laptop to make sure everything is clean. Then I executed the test twice. The first time was successful, but the second time fails:

[03_7032b5e.log](https://github.com/user-attachments/files/17109433/03_7032b5e.log)

Could it also be that the previous run did not close or delete the lock files properly?

SzymonHitachi, 2024-09-25:
Is there any PR for those changes to look at? Especially with the fact that questasim introduced `qrun` command that wraps all the other commands in one step

LarsAsplund, 2024-09-25:
@SzymonHitachi You can find the work in the https://github.com/VUnit/vunit/tree/three-step-flow branch. There is a simple testbench https://github.com/VUnit/vunit/tree/three-step-flow/examples/vhdl/three_step_flow which we use to get a first proof of concept. It works for me but not for @tasgomes and I'm assuming we have some race conditions.

I'm aware of `qrun` but initially I want to follow a flow that is more generic and applies to all simulators we support. That makes our code cleaner. It should be possible to do what is done in `qrun` though. Apart from making Questa run in a multi-threaded setup, we are also planning to add compile groups which allow for compiling files in groups. I know that Mikael prefers that approach to compilation but there are more use cases we need to support so we need to keep our design flexible. I got an email from Mikael today (no coincidence I assume :smile:). I'm hoping he will join the discussion here as he knows Questa under the hood.

LarsAsplund, 2024-09-25:
What I think we need from Siemens at this point is:
1. Under what circumstances can two `vopt` runs be considered independent such that they can run concurrently.
2. At what point can VUnit assume that `vopt` has "released" all files.

SzymonHitachi, 2024-09-25:
> @SzymonHitachi You can find the work in the https://github.com/VUnit/vunit/tree/three-step-flow branch. There is a simple testbench https://github.com/VUnit/vunit/tree/three-step-flow/examples/vhdl/three_step_flow which we use to get a first proof of concept. It works for me but not for @tasgomes and I'm assuming we have some race conditions.
> 
> I'm aware of `qrun` but initially I want to follow a flow that is more generic and applies to all simulators we support. That makes our code cleaner. It should be possible to do what is done in `qrun` though. Apart from making Questa run in a multi-threaded setup, we are also planning to add compile groups which allow for compiling files in groups. I know that Mikael prefers that approach to compilation but there are more use cases we need to support so we need to keep our design flexible. I got an email from Mikael today (no coincidence I assume 😄). I'm hoping he will join the discussion here as he knows Questa under the hood.

Thanks for the links. It seems it still needs modelsim setup, so I guess it requires to have either modelsim/questasim installed or some ENV var defined to use one or another?

tasgomes, 2024-09-25:
@SzymonHitachi Currently, VUnit does not distinguish Modelsim from Questa. To use Questa you need to define `modelsim` as simulator and then provide the path to the Questa installation folder. For instance:

```python
environ["VUNIT_SIMULATOR"] = "modelsim"
environ["VUNIT_MODELSIM_PATH"] = "C:/intelFPGA_pro/21.2/questa_fe/win64"
```

outdoorsweden, 2024-10-04:
Mikael Andersson, Siemens EDA here!
I have not read the whole thread. I've just browsed through it.
I'd recommend to use the **three step flow** (compile, optimize, simulate).

Here you need to make some choices when it comes to optimizations. The alternatives are:

- Optimize each test. Pro: Best performance on each simulation. Drawback: You might get issues with locks when multiple vopt are started at the same time.  But this can be handled by using the undocumented switch "-nolock" to vopt and by making sure that the output from vopt has a unique name. So for each test it would run:

```
>vopt tb -nolock -g <test generics> -o tb_<testname>_opt -debug -designfile design_<test>.bin
>vsim tb_<testname>_opt -qwave=+signal+msg+assertion=pass ....

```

- Create one common optimized version. Pro: Just one optimization. Drawback: Could affect performance significantly if 
  you apply floatgenerics to aggressive. Also you will need to use a hybrid elaboration flow for Visualizer (Not hard).
  
```
>vcom .... 
>vopt tb -floatgenerics runner_cfg -g <default test generics> -o tb_dbg_opt -debug -designfile design.bin

```
   And then foreach test:
```
>vsim tb_dbg_opt -g <test generics> -qwave=+signal+msg+assertion=pass ....

```

You can also generate different optimizations with vopt. Like in the second case, you could generate one for performance in regression only and one for debug:
```
>vcom .... 
>vopt tb -floatgenerics runner_cfg -g <default test generics> -o tb_dbg_opt -debug -designfile design.bin
>vopt tb -floatgenerics runner_cfg -g <default test generics> -o tb_opt 

```
And if you want best performance, you use the tb_opt version without logging:
foreach test:
```
>vsim tb_opt -g <test generics> ....

```

Hope this is helpful!
BR
Mikael
Siemens EDA
Siemens Digital Industries Software

Evenemangsgatan 21
SE-169 03 Solna, Sweden
Mobile: +46 70 932 9516
[andersson.mikael@siemens.com](https://github.com/VUnit/vunit/issues/andersson.mikael@siemens.com)
[www.sw.siemens.com](http://www.sw.siemens.com/)

outdoorsweden, 2024-10-10:
Mikael Andersson, Siemens EDA here again!
I have now used our tools to show the effect of using qrun + the flow "Create one common optimized version" described above.

I have used the axi_dma example. The only change is that the "Random AXI configuration" test has been extended to run a bit longer.  And I choose to measure the effect on a "clean" start. The way you would do in a Continuous Integration environment.

This picture is a screenshot from Questa Run Manager where I defined a flow that is suitable for regression of Vunit based testbenches:
[image]
Notice that I compile and optimize for each testbench. Since the compile step is using qrun, the compile step is so fast that it really does not matter that we do three parallel compile. The fact that we use a library location for each testbench also allowes optimization in parallel without any lock files.
The actual compile and optimize command I used was this:

```
qrun -64 -f .../qrun.files -work axi_dma_lib -optimize -top tb_axi_dma  -snapshot tb_axi_dma_opt 
-vopt.options  -floatgenerics+runner_cfg  +cover=bcesf+axi_dma. -end -outdir .../VRMDATA/my_run/testbench~tb_axi_dma/compile_and_optimize/qrun.out 
```
or from the Makefile:
[image]

And in simulations I used:
```
qrun -64 -work axi_dma_lib -simulate -snapshot tb_axi_dma_opt -vsim.options -f .../tests/Perform_split_transfers.vunit.args 
-coverage -end -onfinish stop -do "coverage save Perform_split_transfers.ucdb -onexit;run -all;exit -f" -t ps -error vsim-3040
```
or from the Makefile:
[image]

So what about performance comparison with current Vunit flow and the flow above? The blue is first run, the orange one is the second run.
[image]

The main reason that multiple cpu's does not make a bigger difference for Vunit is that each simulation does optimization and creates a lock file which next simulation needs to wait until it is removed.

A comparison on only the compile time qrun vs Vunit:
[image]

Hope this is helpful!
BR
Mikael
Siemens EDA
Siemens Digital Industries Software

Evenemangsgatan 21
SE-169 03 Solna, Sweden
Mobile: +46 70 932 9516
[andersson.mikael@siemens.com](https://github.com/VUnit/vunit/issues/andersson.mikael@siemens.com)
[www.sw.siemens.com](http://www.sw.siemens.com/)

LarsAsplund, 2024-10-11:
Hi @outdoorsweden,

> The main reason that multiple cpu's does not make a bigger difference for VUnit is that each simulation does optimization and creates a lock file which next simulation needs to wait until it is removed.

VUnit currently runs vopt as part of the internal "simulation" step as it is a simpler first modification of the current VUnit structure. It is still reusing previous vopt runs though. If we have a testbench running 5 times (with different generics), there will only be one vopt run. That is visible in the debug logs of @tasgomes tests.

Currently, I wait for the lock file to be removed before releasing the thread such that a new simulation can begin. That can be improved by checking for lock files before beginning a simulation instead. If the next simulation is towards another library, there will be no wait time at all. Is that what you mean with

> The fact that we use a library location for each testbench also allowes optimization in parallel without any lock files.

Or are you actively using the `-nolock` feature?

Regarding the difference between the 1 CPU run and the 5 CPU run. If I interpret your measurements correctly, there is no difference between qrun and VUnit for the simulation runs. In both cases, the orange bars becomes 24 time units faster in the 5 CPU case. In the best of worlds, the 5 CPU run would be 5x faster but in short tests like these, the simulator startup time becomes dominant.

I think you've confirmed that simultaneous vopts on the same library isn't possible but how deep does an optimization go? If testbench A and B are optimized towards **different** libraries but they both use module C, will they both try to optimize C? Or is vopt limited the the top-level and whatever C design already existing (optimized or not optimized) is the one being used?

>

outdoorsweden, 2024-10-11:
> VUnit currently runs vopt as part of the internal "simulation" step as it is a simpler first modification of the current VUnit structure. It is still reusing previous vopt runs though. If we have a [...]

This does indeed look like a race condition.  I have never seen anything like it. How do you check that vopt has finished before you start vsim?  Because what you normally would do is:

1. compile
2. optimize
3. Launch simulations in parallel when optimize has finished
If you can recreate the problem, file a SR on https://support.sw.siemens.com/en-US/ and provide the test case.

> Or are you actively using the `-nolock` feature?
No , it is undocumented and does not work the way I expected it.

> > The fact that we use a library location for each testbench also allowes optimization in parallel without any lock files.

So this is the directory structure that I get with Questa Run Manager (I have filtered away some stuff):
[image]

So each testbench have a qrun.out directory that contains all the libraries.  And this is why I can run all the optimizations in parallel. 

> I think you've confirmed that simultaneous vopts on the same library isn't possible but how deep does an optimization go? If testbench A and B are optimized towards **different** libraries but they [...]

No, so all the machine code generated ends up in the optimized version.   So if you want to optimize to different libraries, this might work (have not tested if it generates a lock file  in the design lib or not:

[image]

I will update my Questa vrun application so that it tests this concept.

BR
Mikael

LarsAsplund, 2024-10-12:
@outdoorsweden Just to be clear. There **were** race conditions that we fixed and none of us experience any problems at this point. However, since we weren't sure about the inner workings it was hard to be fully confident that it would work for everyone.

Initially I used OS synchronization mechanisms to make sure that a vopt operation performed by one thread returns before another thread starts doing a new vopt in the same lib. A mistake in that code was the cause of one race condition.

That approach was not enough since the lock file may remain on the file system after the vopt call has returned. Probably due to delays in the file system. A second vopt to the same lib may see that lock file before it is deleted and then it fails. That was the cause of the other race condition we've seen. It was fixed by also checking for the presence of the lock file and wait for it to disappear before letting the next thread run.

I don't think checking the lock file alone is enough. I've observed that vopt can create and delete a lock file several times during its execution. For that reason, VUnit will use an OS lock to prevent other threads from starting a vopt on a lib already used by another vopt. That OS lock is removed when the first vopt call returns **and** there are no remaining lock files.

This means that we do follow your suggestion of 1. compile, 2. optimize, and 3 simulate for each testbench. However, step 2 and 3 are performed in several concurrent threads. No constraint are placed on what can be simulated in parallel. The only constraint now is that no concurrent vopt is allowed on the same lib.

We discussed compiling all libraries into multiple directories or simply make multiple copies of the original set of compiled libs. However, that doesn't feel like a nice solution considering we can have hundreds of testbenches that need their own library copy. I much rather give up the idea of concurrent vopts and have a single library set.

Optimizing each design to a separate lib sounds more interesting as it doesn't involve multiple copies. I will try that and see what lock files are being generated.

Final question about vopt. I feel I don't quite understand the basics. What is being optimized? From what I understand we only call vopt on the testbench (test_counter in your example) but never on the design being tested (the counter)

LarsAsplund, 2024-10-14:
@outdoorsweden I made a quick test where the design is optimized into another directory. It looks like the lock files only appear in the new directory. I think it would work as a workaround if the current solution, which seems to work right now, proves to have some yet to be seen issues.

One issue that remains though is the issue with the slow test as discussed earlier in this thread:

[image]

Sometimes a random test case takes much longer time than it should but sometimes all tests run as expected. This problem is present with multiple threads even if I completely disable optimization. I looked briefly at it before and concluded that it is vsim that adds the extra time. Do you have any idea why that is? Currently all threads running vsim is doing that from the same working directory. I recall we have had discussions about that in the past. Is that a problem? Should all threads run in separate directories?

outdoorsweden, 2024-10-14:
> Final question about vopt. I feel I don't quite understand the basics. What is being optimized? From what I understand we only call vopt on the testbench (test_counter in your example) but never on the design being tested (the counter)

Vopt will optimize testbench and everything in it, including the design.

outdoorsweden, 2024-10-14:
@LarsAsplund 
> Sometimes a random test case takes much longer time than it should but sometimes all tests run as expected. This problem is present with multiple threads even if I completely disable optimization. I [...]
Do you have the example so I can play with it?

LarsAsplund, 2024-10-14:
@outdoorsweden This happens on the dummy testbench used in this thread https://github.com/VUnit/vunit/blob/three-step-flow/examples/vhdl/three_step_flow/tb_example.vhd which we run with 5 different settings for the `value` generic:

``` python
for value in range(5):
    test.add_config(name=f"{value}", generics=dict(value=value))
```

I don't think the test itself is significant so you could also test with the DMA example you had. Run all simulations in parallel in the same directory but remove optimization so that it doesn't play a role.

@tasgomes hasn't experience this so you may not experience anything either. That would suggest that there is another timing depending conflict over shared resources that has to be taken into account. modelsim.ini?

[24 more replies]

closed 2024-12-11 by LarsAsplund