VUnit/vunit #901: GHDL Interface can not detect GCC backend with newer GHDL/LLVM #898

LeFl0w, 2023-02-20:
Hi this error seems related to the issue #898 .   
When trying Vunit using the `ghdl/ghdl:fedora36-gcc-11.3.0` docker image I got the following error.
Everything was working propperly with an older self compiled GHDL gcc.

```
WARNING - /usr/local/lib/python3.10/site-packages/vunit/vhdl/osvvm/ScoreboardPkg_slv_c.vhd: package 'scoreboardpkg_slv' previously defined in /usr/local/lib/python3.10/site-packages/vunit/vhdl/osvvm/ScoreboardPkg_slv.vhd
  ERROR - Could not detect known LLVM backend by parsing 'ghdl --version'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 672, in main
    all_ok = self._main(post_run)
  File "/usr/local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 716, in _main
    all_ok = self._main_run(post_run)
  File "/usr/local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 741, in _main_run
    simulator_if = self._create_simulator_if()
  File "/usr/local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 735, in _create_simulator_if
    return self._simulator_class.from_args(args=self._args, output_path=self._simulator_output_path)
  File "/usr/local/lib/python3.10/site-packages/vunit/sim_if/ghdl.py", line 77, in from_args
    backend=cls.determine_backend(prefix),
  File "/usr/local/lib/python3.10/site-packages/vunit/sim_if/ghdl.py", line 145, in determine_backend
    raise AssertionError("No known GHDL back-end could be detected from running 'ghdl --version'")
AssertionError: No known GHDL back-end could be detected from running 'ghdl --version'
Expected to find one of dict_keys(['mcode code generator', 'llvm (\\d+\\.\\d+\\.\\d+ )?code generator', 'GCC back-end code generator'])
== Output of 'ghdl --version'============================================================
GHDL 3.0.0-dev (v2.0.0-1369-gf04182410) [Dunoon edition]
 Compiled with GNAT Version: 12.2.1 20221121 (Red Hat 12.2.1-
 GCC 11.3.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
``` 

Florent

LarsAsplund, 2023-02-20:
Yes, that seems to be exactly the same. Does it work if you change [this line](https://github.com/VUnit/vunit/blob/393c20c20385742bd72080244096d550508c1cce/vunit/sim_if/ghdl.py#L128) from

``` python
mapping = {
            r"mcode code generator": "mcode",
            r"llvm (\d+\.\d+\.\d+ )?code generator": "llvm",
            r"GCC back-end code generator": "gcc",
        }
```

to

``` python
mapping = {
            r"mcode code generator": "mcode",
            r"llvm (\d+\.\d+\.\d+ )?code generator": "llvm",
            r"GCC (back-end|\d+\.\d+\.\d+) code generator": "gcc",
        }
```

?

LeFl0w, 2023-02-20:
indeed it is working.
thanks.

LarsAsplund, 2023-02-22:
This is now fixed on the main branch and a new release 4.6.2 is being built as we speak. It should be on PyPi shortly.

closed 2023-02-22 by LarsAsplund

tristanitschner, 2024-04-11:
Since ghdl got a JIT backend the issue still occurs. Output of ghdl --version:
```
GHDL 5.0.0-dev (4.0.0.r24.g14b411c57) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I'd suggest using `mcode( JIT)? code generator` regular expression.