VUnit/vunit #909: Feature Request: check_equal(...) overloads for ieee.fixed_pkg.ufixed and ieee.fixed_pkg.sfixed

creiter64, 2023-03-01:
Hi all,

It would be very helpful if VUnit had support for using `check_equal(...)` on the `ieee.fixed_pkg` data types.

Currently I have to do all my testbenches using fixed point arithmetic like this:

```vhdl
check_true(A = B);
```

So that VUnit uses the equality operator from the fixed_pkg.

For verbosity of the test cases it would be better to have:

```vhdl
check_equal(A, B);
```

or even better, if the `check_equal` function could directly convert from the string representation of a fixed point number like this:

```vhdl
check_equal(A, "-13.625");
```

Since the fixed point arithmetic package is part of VHDL-2008 now, I think others might find this useful too.

LarsAsplund, 2023-03-01:
@creiter64 That can be done but I would postpone it until after v4.7. 

Did you know that our `check_equal` subprograms are generated by Python? I would encourage you to have a look at https://github.com/VUnit/vunit/blob/master/vunit/vhdl/check/tools/generate_check_equal.py. There are templates for subprogram APIs, implementations, and tests. The templates have a number of blanks that are specified for each combination of `got` and `expected` parameter types that we support . Maybe you can figure out how to add support for the fix point datatypes? A complicating factor is that this would be the first check equals not supported in VHDL-93 so we will need to create a separate set of files that we use when compiling for VHDL-2008 and beyond. That can be ignored for the first attempt though.

abaebae, 2024-02-23:
I added a PR #989 with `check_equal` support for fixed_pkg types. Any kind of feedback is welcome

LarsAsplund, 2024-03-10:
#989 has been merges.

closed 2024-03-10 by LarsAsplund