VUnit/vunit #910: Feature Request: Option for library by library compilation

abaebae, 2023-03-01:
Currently only file by file compilation is supported. It would be nice to have the option to select a compilation by library, i.e. all files from a library in a single compile command. This is especially useful for SystemVerilog.

LarsAsplund, 2023-03-01:
@abaebae Is your use case library groups only or do you see that arbitrary groups are of interest? If we only need library groups there is no need for a new group concept (or whatever it should be called), only a library configuration controlling what approach to use.

abaebae, 2023-03-02:
@LarsAsplund currently I only see the use case of library groups

LarsAsplund, 2023-03-02:
@abaebae Check. What simulators are you targeting?

abaebae, 2023-03-02:
@LarsAsplund currently only Aldec Riviera-Pro