VUnit/vunit #924: ghdl coverage report fails when base_path of file_name isn't there yet

schrolli, 2023-03-24:
when generating code-coverage reports, the call to `results.merge_coverage()` fails when the supplied parameter `file_name` is a path, where the basepath doesn't exist yet.

In order to not clutter the source-path, i wanted to put all files and folders produced by the gcov/lcov chain into a subfolder of vunit_out.

## how to reproduce
run-script with the call to results.merge_coverage in the 
```
#!/usr/bin/env python3

from vunit import VUnit
import os

def post_run(results):
	merge_path = "vunit_out/coverage/coverage_data"
	#os.makedirs(merge_path, exist_ok = True)
	results.merge_coverage(file_name=merge_path)

VU = VUnit.from_argv()
VU.add_osvvm()

src_files = []

for file in os.listdir('.'):
	if os.path.isfile(file):
		if os.path.splitext(file)[1] == ".vhd":
			src_files.append(file)

LIB = VU.add_library("lib")
LIB.add_source_files(src_files)
LIB.set_compile_option("enable_coverage", True)

VU.set_compile_option("ghdl.a_flags", ["-frelaxed"])
VU.set_sim_option("ghdl.elab_flags", ["-frelaxed"])
VU.set_sim_option("enable_coverage", True)
VU.main(post_run = post_run)
```
## what is expected
when the file_name parameter is simply set to `coverage_data`, this folder is created with no intervention.

it is expected when making the path longer, the basefolders get created, too.

## what is not expected
script exits with assertion error after call to gcov-tool. output ending:
```
===========================================================
pass 9 of 9
===========================================================
Total time was 2.1 seconds
Elapsed time was 2.1 seconds
===========================================================
All passed!
gcov-tool: fatal error: Cannot make directory vunit_out/coverage/coverage_data
compilation terminated.
cannot access directory vunit_out/coverage/coverage_data
cannot access directory vunit_out/coverage/coverage_data
cannot access directory vunit_out/coverage/coverage_data
cannot access directory vunit_out/coverage/coverage_data
cannot access directory vunit_out/coverage/coverage_data
cannot access directory vunit_out/coverage/coverage_data
cannot access directory vunit_out/coverage/coverage_data
cannot access directory vunit_out/coverage/coverage_data
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 672, in main
    all_ok = self._main(post_run)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 716, in _main
    all_ok = self._main_run(post_run)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 761, in _main_run
    post_run(results=Results(self._output_path, simulator_if, report))
  File "./testrun.py", line 9, in post_run
    results.merge_coverage(file_name=merge_path)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/ui/results.py", line 33, in merge_coverage
    self._simulator_if.merge_coverage(file_name=file_name, args=args)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/sim_if/ghdl.py", line 416, in merge_coverage
    assert len(gcda_dirs) == 1, "Expected exactly one folder with gcda files"
AssertionError: Expected exactly one folder with gcda files
```
## changes to mitigate this

The root-cause is in https://github.com/VUnit/vunit/blob/master/vunit/sim_if/ghdl.py#L408 where `gcov-tool` is called with the not-existing path, that it could not handle.

There are several options to change behaviour, that came to my mind:
- catch errors from call to `gcov-tool` and exit appropiately
- check for path in existence and print error message and exit appropiately before the call to `gcov-tool`
- simply create the path beforehand. e.g. `os.makedirs(file_name, exist_ok = True)`

I'm not quite shure which method is preferred, so i wanted to ask before creating a pr.

umarcor, 2023-04-19:
I think that the third method is ok. However, the second would also be acceptable. I don't think it's worth the complexity of the first solution just for solving this issue, because the problem is not with gcov-tool per se. We know the problem is that by design it expects the location to exist, the same as VUnit at the moment.

albydnc, 2023-12-01:
Is there any news on this?

commit 8a5fa2d4 (2024-02-25): Fixes #924
commit bd8a8ba5 (2024-02-25): Fixes #924
closed 2024-02-25 by LarsAsplund