VUnit/vunit #925: Verilator Support

AlexLao512, 2023-04-02:
Is Verilator support possible? With the release of Verilator 5.00x earlier this year, Verilator now supports dynamic scheduling and delay statements making it possible for Verilator to support SystemVerilog testbenches. Antmicro recently did a presentation about plans to eventually support UVM. CocoTB already has support for Verilator as well but I much prefer the VUnit workflow.

umarcor, 2023-04-28:
Since the main file all other SV files in this repository depend on is https://raw.githubusercontent.com/VUnit/vunit/master/vunit/verilog/vunit_pkg.sv, I did the following test:

```bash
#!/usr/bin/env bash

set -e

cd $(dirname "$0")

mkdir -p verilator-vunit-test
cd verilator-vunit-test
wget https://raw.githubusercontent.com/VUnit/vunit/master/vunit/verilog/vunit_pkg.sv

verilator="docker run --rm --security-opt label=disable -itv $(pwd):/wrk -w /wrk gcr.io/hdl-containers/verilator verilator"

$verilator --version

set +e
$verilator --binary -j 0 -Wall vunit_pkg.sv
set -e

cd ..
rm -rf verilator-vunit-test
```

It's currently failing as:

```
%Error-UNSUPPORTED: vunit_pkg.sv:222:13: Unsupported: disable fork
  222 |             disable wait_or_timeout;
      |             ^~~~~~~
%Error-UNSUPPORTED: vunit_pkg.sv:226:13: Unsupported: disable fork
  226 |             disable wait_or_timeout;
      |             ^~~~~~~
```

The full log is the following:

```
--2023-04-27 13:51:08--  https://raw.githubusercontent.com/VUnit/vunit/master/vunit/verilog/vunit_pkg.sv
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.108.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6687 (6.5K) [text/plain]
Saving to: 'vunit_pkg.sv'

vunit_pkg.sv                    100%[=======================================================>]   6.53K  --.-KB/s    in 0s      

2023-04-27 13:51:08 (31.5 MB/s) - 'vunit_pkg.sv' saved [6687/6687]

Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
Verilator 5.009 devel rev v5.008-150-gee5c0a290
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
%Warning-DECLFILENAME: vunit_pkg.sv:9:1: Filename 'vunit_pkg' does not match CLASS name: 'test_runner'
    9 | class test_runner;
      | ^~~~~
                       ... For warning description see https://verilator.org/warn/DECLFILENAME?v=5.009
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Warning-VARHIDDEN: vunit_pkg.sv:142:17: Declaration of signal hides declaration in upper scope: 'exit_without_errors'
  142 |       int       exit_without_errors;
      |                 ^~~~~~~~~~~~~~~~~~~
                    vunit_pkg.sv:24:17: ... Location of original declaration
   24 |    int          exit_without_errors = 0;
      |                 ^~~~~~~~~~~~~~~~~~~
%Error-UNSUPPORTED: vunit_pkg.sv:222:13: Unsupported: disable fork
  222 |             disable wait_or_timeout;
      |             ^~~~~~~
%Error-UNSUPPORTED: vunit_pkg.sv:226:13: Unsupported: disable fork
  226 |             disable wait_or_timeout;
      |             ^~~~~~~
%Error: Exiting due to 2 error(s), 2 warning(s)
```

There is an issue by the maintainer of SVUnit, opened a couple of weeks ago:

- verilator/verilator#4125

It seems there is work in progress to allow process execution control, which would allow to implement `disable fork`.
However, at the moment, verilator does not support enough non-synthesisable features.
We should revisit this in some weeks/months, according to https://github.com/verilator/verilator/issues/4125#issuecomment-1511167069.

On the other hand, I don't know whether the temporary approach used in SVUnit can be used in VUnit: https://github.com/svunit/svunit/commit/6093f363f2a1b634f994db8f212594b49f0034ec.

svenka3, 2023-06-28:
I would be interested too - try and mask disable fork for now and see if this helps. Any comments anyone please?

LarsAsplund, 2023-07-01:
I think the workaround SVUnit is using would be acceptable for VUnit as well if we finish Verilator support before Verilator supports `disable`.

umarcor, 2023-07-02:
FTR, there was an update in verilator/verilator#4125: https://github.com/verilator/verilator/issues/4125#issuecomment-1572146533

> one could implement `disable fork` by getting the `std::process` instance of each forked process and calling `kill` on them. Or by extending the process control infrastructure with a concept like process trees (this may be better as it avoids problems with fork lifetimes/capture semantics).

rdvdev2, 2024-10-13:
As an update to the fork situtation, while `disable fork` is now supported in verilator, `disable fork by name` isn't. There is a tracking issue here https://github.com/verilator/verilator/issues/5432, and following are my logs trying the script by @umarcor:

```
Verilator 5.027 devel rev v5.026-118-gec0815e9a
%Warning-DECLFILENAME: vunit_pkg.sv:9:1: Filename 'vunit_pkg' does not match CLASS name: 'test_runner'
    9 | class test_runner;
      | ^~~~~
                       ... For warning description see https://verilator.org/warn/DECLFILENAME?v=5.027
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Warning-VARHIDDEN: vunit_pkg.sv:142:17: Declaration of signal hides declaration in upper scope: 'exit_without_errors'
  142 |       int       exit_without_errors;
      |                 ^~~~~~~~~~~~~~~~~~~
                    vunit_pkg.sv:24:17: ... Location of original declaration
   24 |    int          exit_without_errors = 0;
      |                 ^~~~~~~~~~~~~~~~~~~
%Error-UNSUPPORTED: vunit_pkg.sv:222:13: Unsupported: disabling fork by name
  222 |             disable wait_or_timeout;
      |             ^~~~~~~
%Error-UNSUPPORTED: vunit_pkg.sv:226:13: Unsupported: disabling fork by name
  226 |             disable wait_or_timeout;
      |             ^~~~~~~
%Error: Exiting due to 2 error(s), 2 warning(s)
```

rdvdev2, 2024-10-14:
I could get around this by replacing the named disables with a `disable fork`. Another thing that needed to be changed was the `TEST_SUITE_FROM_PARAMETER` macro, as verilator doesn't like `__runner__.loop`, it had to be `__runner__.loop()` instead. With this, the system verilog side of things compiles.

On the other hand, there are a couple of blockers on the verilator side:
- Files can't be compiled individually, full compilations must be triggered each time a file changes (as far as I can tell, or as far as my ability with the tool reaches)
- There's no notion of libraries except DPI libraries, which can't be used with the `--timing` flag (https://veripool.org/guide/latest/exe_verilator.html#cmdoption-lib-create), I'm not sure if there's some workaround that can be used to work with the libraries defined in vunit
- A different compilation is needed for each top that is used
- Generics for the top are defined at compile time, meaning that each different `runner_cfg` requires a new compilation

All of this is to say that verilator requires lots of recompilations, and this can be a pretty heavy cost in big projects. An idea that comes to mind is to modify `TEST_SUITE` or `TEST_SUITE_FROM_PARAMETER` (whichever fits) to take the `runner_cfg` from the arguments passed to the simulator (this would reduce the amount of recompilations). If all possible tops could be instantiated by a vunit-generted top that delegates execution at runtime, we could possibly get away with a single compilation per test suite run (opposed to a compilation per test bench). As to rebuilding single files, verilator can handle it with the generated Makefile, but work must be done in vunit to support external handling of the build graph. All of this should be done in a way that keeps all other simulators compatible, be it by keeping the current behavior or be it by adopting the new single-top, command-line controlled, architecture).

This is my analysis after playing with the code for half a day, but I'm no expert in either tool, and I would love to hear what the project maintainers think about this and what could be the course of action to get verilator working with vunit.

(If anything in this message needs clarification/expansion, please ask. Here it currently is 2 AM, and I'm trying my best to share my findings before I go to sleep and forget half of it.)

TLDR: It looks like adding support to verilator is a really non-trivial task due to the verilator's approach to compilation of verilog code. Opinion from maintainers would be appreciated.

LarsAsplund, 2024-10-14:
Thanks for the investigation @rdvdev2! Here are my initial reflections

> I could get around this by replacing the named disables with a `disable fork`. Another thing that needed to be changed was the `TEST_SUITE_FROM_PARAMETER` macro, as verilator doesn't like [...]

I can try what the implications are with the other supported simulators.

> 
> On the other hand, there are a couple of blockers on the verilator side:
> 
> * Files can't be compiled individually, full compilations must be triggered each time a file changes (as far as I can tell, or as far as my ability with the tool reaches)

There is a planned work to introduce compilation groups which is a concept that groups files that must be compiled with the same compile call. Grouping all files would then be a way forward for Verilator. Maybe not the most efficient approach but a good first step.

> * There's no notion of libraries except DPI libraries, which can't be used with the `--timing` flag (https://veripool.org/guide/latest/exe_verilator.html#cmdoption-lib-create), I'm not sure if there's some workaround that can be used to work with the libraries defined in vunit

If all files belong to the same group, there can only be a single library.

> * A different compilation is needed for each top that is used
> * Generics for the top are defined at compile time, meaning that each different `runner_cfg` requires a new compilation
> 

There has been discussions to support a file-based runner_cfg as that may allow more aggressive optimizations by some simulators. An implementation for VHDL exists in a branch but the same concept could be ported to SystemVerilog

> All of this is to say that verilator requires lots of recompilations, and this can be a pretty heavy cost in big projects. An idea that comes to mind is to modify `TEST_SUITE` or [...]

Maybe compilation can be initiated by the run script but managed completely independent of VUnit. Maybe the resulting artifact can then be imported and treated as an external library

rdvdev2, 2024-10-22:
Thanks for the insights @LarsAsplund, is there any relevant PR or issue related to the changes you mentioned where I could help to get closer to getting this working?

LarsAsplund, 2024-10-22:
[This](https://github.com/VUnit/vunit/commit/c30232998448f07ad951d646399dbf9db34c5e1f) is the commit that introduces file-based `runner_cfg`s. It is part of the branch that was cancelled since we found a better solution to the use case initiating that branch. However, the commit can be cherry-picked to the main branch in order to serve other use cases.

#910 is the issue partly touching the compile group concept.