VUnit/vunit #957: package "tests" in Python path?

dodotronix, 2023-09-18:
Vunit seems to install its tests into "test" in the Python path.

I found out that, if i create my own archlinux package according to the pkgbuild guidelines from archwiki, i get following error when i run makepkg -si

```
error: failed to commit transaction (conflicting files)
vunit-hdl-git: /usr/lib/python3.11/site-packages/tests/__init__.py exists in filesystem (owned by python-protonvpn-nm-lib)
vunit-hdl-git: /usr/lib/python3.11/site-packages/tests/__pycache__/__init__.cpython-311.opt-1.pyc exists in filesystem (owned by python-protonvpn-nm-lib)
vunit-hdl-git: /usr/lib/python3.11/site-packages/tests/__pycache__/__init__.cpython-311.pyc exists in filesystem (owned by python-protonvpn-nm-lib)

```

# Here is my package
```
#ARCH PKGBUILD

_gitrepo='vunit'
pkgname='vunit-hdl-git'
pkgver=4.7.0.40
pkgrel=1
pkgdesc="Vunit framework for writing automated HDL testbenches"
arch=('any')
url="https://github.com/VUnit/vunit.git"
license=('GPL3')
makedepends=('python-build' 'python-installer' 'python-setuptools-scm' 'python-setuptools' 'python-wheel')
depends=('python-colorama')
conflicts=($pkgname)
source=("git+https://github.com/VUnit/$_gitrepo.git")
sha384sums=('SKIP')

pkgver() {
  cd "$srcdir/$_gitrepo"
  sed -i 's/\(setuptools_scm\).*/\1 >= 2.0.0"/g' pyproject.toml
 git describe --tags --long | sed 's/^v\([^-]*-[^-]*\)-g.*/\1/;s/-/./g'
}

build() {
  cd "$srcdir/$_gitrepo"
  python -m build --wheel --no-isolation
}

package() {
  cd "$srcdir/$_gitrepo"
  python -m installer --destdir="$pkgdir" dist/*.whl
}
```

According to the [archwiki](https://wiki.archlinux.org/title/Python_package_guidelines) in the chapter "Test directory in site-package" it is not recommended practice to install tests in __/usr/lib/pythonX.Y/site-packages/tests/__

LarsAsplund, 2025-08-11:
My plan is to remove test code from the installation and only keep it for those cloning the repo.

glencoe, 2025-08-12:
#1130 solves the issue in our case. Not sure though if that's how you'd want to handle this.

closed 2026-01-07 by LarsAsplund