VUnit/vunit #960: Feature: axi master implementation

DavidMartinPhios, 2023-09-28:
Hi guys, 

thank you for the open source project. We are using the project in several of our projects and we are really satisfied with it. 
We are also heavily using the verification components especially in combination with AXI interfaces.   

What we are missing is a full axi master verification component. Is there a plan from your side to implement this component or is there a special reason why you are not implement it?

If this feature is not planned from your side we would really like to support the implementation of  this feature with our development team and would be pleased to get in direct contact for discussion. 

Thank you.

CodablePanda, 2023-11-14:
I'm also interested and would be happy to help out with this.
I have been looking at the boards and see that some comments indicate that there is progress in rewriting the verification components (#726 and #480 for example).
I couldn't find the repository where this work is happening though.
Is there an update on the status of the verification components and is there a way for me to contribute?

DavidMartinPhios, 2024-05-06:
Hi, after a longer break from my side i would like to start with the implementation of the axi master verification component. Are there any updates related to this component or to any other rewriting progress?

willkamp, 2024-08-04:
I've had a go at a Vunit AXI master, which serves my purpose reasonably well. Several processes, one for each of the AXI channels, that are coordinated by message queues. I would be very happy to contribute it as a starting point.