VUnit/vunit #976: Feature Request: Support for adding .psl files to allow for Assertion Based Verification

vhdlp, 2023-11-09:
Hi everybody,

I've been using VUnit for a few years now. It is a really good tool especially due to the simulator independence. However, it would be really good if one could add not only pure RTL source files, but also e.g. `.psl` files required for Assertion Based Verification. Sure, it is possible to add the PSL code directly within VHDL (and I already did that), but in a new project it is a requirement to store them separately.

At least in RivieraPro (in Questa it is the same I think) it is required to compile the source file and the corresponding `.psl` file within the same compilation run, i.e. the same `vcom` call, which requires that e.g. `file.vhdl` and `file.psl` are somehow grouped together. Currently I do not see any possibility to reach this using VUnit, at least as long as VUnit has control over the corresponding sources. 

Do you see any possibility to add such a feature? So maybe adding associated files by adding source files like `add_source_file(..., associated_files=[('psl', 'file.psl'), ...])`.

Best regards
Peter

vhdlp, 2023-11-09:
Some more info: I already tried to use `set_compile_option()` of the `SourceFile` object returned by `add_source_file(...)` and specify e.g. `set_compile_option("rivierapro.vcom_flags", ["-psl", "file.psl"])`, but unfortunately this does not work, because of wrong order of parameters passed to `vcom`.

LarsAsplund, 2023-11-19:
This is one of several issues that basically points in the same direction. Fairly high on the priority list is to add a "compilation group" concept which basically is a list of source files compiled together