VUnit/vunit #980: GHDL fails with exit code 4294967295

hcommin, 2023-12-12:
This minimal VUnit testbench succeeds with Modelsim, but fails with GHDL. However, when I try to reproduce the error with standalone GHDL (without VUnit), it seems to work fine.

Here is the minimal VUnit testbench:
```vhdl
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

library vunit_lib;
    context vunit_lib.vunit_context;

entity tb is
    generic(
        runner_cfg : string
    );
end tb;

architecture sim of tb is
    
    type SlvArray_t is array(integer range<>) of std_logic_vector;
    type UnsignedArray_t is array(integer range<>) of unsigned;
    
    constant a : UnsignedArray_t(0 to 0)(0 downto 0) := (others => (others => '0'));
    constant b : SlvArray_t := SlvArray_t(a);
    
begin
    
    process
    begin
        test_runner_setup(runner, runner_cfg);
        
        while test_suite loop
            if run("test") then
            end if;
        end loop;
        
        test_runner_cleanup(runner);
        wait;
    end process;
    
end sim;
```

It produces this output, indicating return code 4294967295:

```
$ python run.py --simulator=ghdl --simulator-path=$GHDL_PATH --log-level debug

...

  DEBUG - Waiting for process with pid=8356 to stop
  DEBUG - Process with pid=8356 terminated with code=4294967295
fail (P=0 S=0 F=1 T=1) lib.tb.test (1.9 seconds)
```

This is my attempt at a standalone testbench:
```vhdl
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

entity tb is
end tb;

architecture sim of tb is
    
    type SlvArray_t is array(integer range<>) of std_logic_vector;
    type UnsignedArray_t is array(integer range<>) of unsigned;
    
    constant a : UnsignedArray_t(0 to 1)(7 downto 0) := (others => (others => '0'));
    constant b : SlvArray_t := SlvArray_t(a);
    
begin
    
    process
    begin
        std.env.stop;
    end process;
    
end sim;
```

This seems to run correctly with GHDL and return code 0 (EXIT_SUCCESS):
```
$ ghdl -a --std=08 tb.vhd
$ ghdl --elab-run --std=08 tb
simulation stopped @0ms
$ echo $?
0
```

LarsAsplund, 2023-12-14:
Running this myself with Python 3.12.0, Windows 10 and

```
GHDL 4.0.0-dev (3.0.0.r636.g1ec1eb6d9) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode code generator
```

results in a pass. I think this is related to what build and version you're using for GHDL so I suggest that you open an issue on their side.

hcommin, 2024-02-26:
@LarsAsplund No, I was not able to resolve it and I had to drop GHDL so I could move forward with my project. It's a shame because GHDL is really great when it works. (Unlimited parallel instances with no license fees).

LarsAsplund, 2024-02-26:
Ok, in that case I will close this issue. As an alternative you can also try the open source NVC simulator. That will also give you unlimited number of licenses.