VUnit/vunit #981: ModelSim issues warning regardingstring_ops.vhd

nselvara, 2023-12-19:
Hello, when I run VUnit test the ModelSim spits out a warning regarding the `string_ops` 
The warning is: `\vunit\vhdl\string_ops\src\string_ops.vhd(565): (vopt-1083) Implicit array operator "=" always returns FALSE (left length 1 is not equal to right length 0).`

The flagged line of code is this:
[image]

I think the if clause can be rewritten to:
```vhdl
if substring'length = 0 then
    return start_pos;
end if;
```

LarsAsplund, 2024-02-25:
Failed to reproduce this myself with my versions of Modelsim and Questasim. Have you verified that the warning goes away with the suggested change? If it thinks the substring has a length of 1 the substring'length = 0 would also always be false.

nselvara, 2024-02-28:
Hi, thx for reply. I now checked again and also replaced with the suggested solution. The warning for real disappears. I tried with both variants. I'd guess `1` is probably the correct one, as the string type itself is at least of length 1.

LarsAsplund, 2024-02-28:
The if statement covers a corner case in the `find` function: What should it return when we search for the empty string? Just like the `find` method in Python, we return 0 in this case.

The `find` function comes in two variants: One searching for input parameter `substring` of type `string` and one searching for input parameter `char` of type `character`. The former can have the empty string as input which motivates the if statement. The latter creates a one-character string of the input `char` and then calls the former. In that case the input `substring` to the former will always be of length 1. 

```
  function find (
    constant s : string;
    constant char : character;
    constant start : natural := 0;
    constant stop : natural := 0)
    return natural is
    variable substring : string(1 to 1);
  begin
    substring(1) := char;
    return find(s, substring, start, stop);
  end;
```

For some reason Questa forgets that the function searching for a string can be called directly. I notice that the `string` version is declared before the `character` version but is defined after. Does it make a difference if the definition of the `string` version is placed first?

nselvara, 2024-02-29:
Good morning, I see it seems to be an order issue or some kind of like that.
Did you mean like this:

```vhdl
function find (
    constant s : string;
    constant char : character;
    constant start : natural := 0;
    constant stop : natural := 0)
    return natural is
    variable substring : string(1 to 1);
  begin
    substring(1) := char;
    return find(substring, s, start, stop);
  end;

  function find (
    constant substring : string; -- Placing this at the top, right?
    constant s : string;
    
    constant start : natural := 0;
    constant stop : natural := 0)
    return natural is
    variable start_pos, stop_pos : natural;
    variable o : natural;
  begin
    if start = 0 or left_of_range(s, start) then
      start_pos := s'left;
    elsif right_of_range(s, start) then
      return 0;
    else
      start_pos := start;
    end if;
```

This doesn't change anything. I still get the warning. It seems to be that the attributes are probably not evaluated in the static analysis of QuestaSim. 

Apropos warnings, I also get this but only when I compile it 😅: 
`** Warning: C:\Repo\delta_firmware\.venv\Lib\site-packages\vunit\vhdl\data_types\src\id_pkg.vhd(351): (vcom-1246) Range 1 to 0 is null.`

```vhdl
impure function get_lineage(id : id_t) return id_vec_t is
    impure function get_lineage_i(id : id_t) return id_vec_t is
      variable parts : lines_t := split(full_name(id), ":");
      constant length : positive := parts'length + 1;
      variable lineage : id_vec_t(1 to length);
      variable tmp_id : id_t := id;
    begin
      for idx in length downto 1 loop
        lineage(idx) := tmp_id;
        exit when tmp_id = root_id;
        tmp_id := get_parent(tmp_id);
      end loop;

      return lineage;
    end;
  begin
    if id = null_id then
      null_id_failure("get_lineage");
      return (1 to 0 => null_id);  -- This one causes the warning
    elsif id = root_id then
      return (1 => root_id);
    end if;

    return get_lineage_i(id);
  end;
```

LarsAsplund, 2024-03-03:
No, I was thinking switching the order of the complete function definitions such that it matches the order the functions were declared. Also, the function depending on the other comes last

nselvara, 2024-03-04:
Ahh, now I see, you meant the order of the definition and the declaration. Sry, for misunderstanding it. 
I give it a try and report back.

closed 2024-03-09 by LarsAsplund

rafaelnp, 2024-06-05:
The same issue is arising again, the line 488 has the same if statement ([string_ops.vhd#L488](
https://github.com/VUnit/vunit/blob/2c0b75e40aa59a42c51b5f6b2c40767f1b2b83d4/vunit/vhdl/string_ops/src/string_ops.vhd#L488))branch and got a signal 11 when using questa on Linux:

```sh
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: /home/user/src/vunit/vunit/vhdl/string_ops/src/string_ops.vhd(488): (vopt-1083) Implicit array operator "=" always returns FALSE (left length 1 is not equal to right length 0).
# vopt_stacktrace.vstf written
# ** Fatal: Unexpected signal: 11.
# ** Note: /**/**/**/**_tb.vhd(203): Vopt Compiler exiting
# ** Error: (vopt-2064) Compiler back-end code generation process terminated with code 232.
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=2, Warnings=1.
# Error loading design
Error loading design
# End time: 00:30:41 on Jun 06,2024, Elapsed time: 0:00:01
# Errors: 2, Warnings: 1
```

Questa version:
[image]

commit c68626dc (2024-03-09): Removed Questasim compilation warning. Fixes #981.
commit 315adae5 (2024-06-10): Removed Questasim compilation warning. Fixes #981. (2)